kakopappa/sinric #184: Devices Not Responding

GaryLittlemore, 2019-01-01:
My two devices have stopped working and both are saying “Not responding”, is there a problem?

MacSass, 2019-01-01:
Hi @GaryLittlemore , @kakopappa ,
I can confirm this - all of my devices stopped responding - as much as I can say since the new year ...

I checked the ESP side and ESP does not receive any requests from sinric ...
I think there is an issue with the service currently.

Can others confirm this? Anyone still has working switches / lamps?

kakopappa, 2019-01-02:
Thank you for letting me know.

I am on the way to work now. I will fix this when I go back home tonight

Please use the old version of websocket library for now.

Thanks and happy new year to you all

dhpete, 2019-01-02:
Yep, same here. Devices not responding with Alexa voice commands, but still able to use the switch in the App.

MacSass, 2019-01-02:
@dhpete : Which app? Alexa app? Does not work as well for my devices ...

@kakopappa : I think the current issue of all devices not working anymore is not regarding the websocket version ... I did not change anything from last year and it just stopped working. 
I though it might be my ESP, and then found the old code does not compile anymore (due to updated library), but the root cause of all devices not working right now must be somewhere in the backend I think ... I don´t think that this can be fixed on our end ...

Regards - MacSass

kakopappa, 2019-01-02:
Hi guys,

This issue is related to ios alexa app skill linking issue fix I made a few days ago. When Alexa renews the oauth token, i have sent an invalid token back to alexa. Now when you ask Alexa to do something it's sending a request with an invalid token and server has no idea to which account is making the request

unfortunately, to fix this it requires to link the re-link (disable/enable) the skill so it will send a valid token next time.

MacSass, 2019-01-02:
Hi @kakopappa ,
I can confirm that after disabling, then re-enabling and re-linking the account everything is working again ...

Thanks for the support - MacSass

dhpete, 2019-01-02:
Yes, confirmed working again. Still kinda sketchy with voice commands, but
Alexa can be that way sometimes. Thanks for your hard work.

gizmocuz, 2019-01-03:
https://github.com/kakopappa/sinric/issues/185

GaryLittlemore, 2019-01-03:
I’m all up and runnning again, great work Kakopappa

closed 2019-01-03 by GaryLittlemore

malhotrasujeet, 2019-01-03:
Same problem. Solved my Disabling/ Re-enabling the Skill as suggested. Thanks Kakopappa for the awesome work!

MacSass, 2019-01-03:
Hi @kakopappa ,
after having the skill re-linked yesterday it was working again, but failed again today.
I tried once again to re-link (disable skill / re-enable skill in Alexa app), but after entering my sinric credentials alexa app tells me that the skill currently can not be linked ... I should try again later.

Are you positive that the isse has been fixed permanently?
Anybody else having another failure after re-linking?

Thanks - MacSass

kakopappa, 2019-01-04:
I have rolled-back the update. I restored the previous working version
again.

Disable skill / re-enable skill in Alexa app was the issue I tried to fix.

use a PC and Chrome. It will work fine.

MacSass, 2019-01-04:
Hi @kakopappa ,
thanks - I can confirm it is working again after re-linking with Google Chrome from http://alexa.amazon.de .

I could not get the skill activated within the iOS Alexa app though (this was working before).

I let you know if it is working stable now ...

Thanks - MacSass