kakopappa/sinric #185: Devices Not Responding

6575bala, 2019-01-02:
My TV,CABLE BOX devices have stopped working and both are saying “Not responding”, is there a problem?

kakopappa, 2019-01-02:
Hi. There seems to be a problem with the server at the moment.. Seems to be related server crashing and restarting when someone connect using the new Arduino Websocket library. I am at work. I will check when I go back home in few hours

amitabh-srivastav, 2019-01-02:
Today I faced same problem for all my Sinric devices. The solution suggested by @kakopappa against thread#28 worked fine and the problem got solved. All devices updated their status on Alexa App and are working fine. I am reproducing the solution given in #28:

1. Unlink the skill in your Alexa app,
2. Link the skill again with the same account you used to create devices in sinric.com
3. Scan for devices.

Dhanush4444, 2019-01-02:
@amitabh-srivatsav change your devices name to which are easy to spell. Sometimes Alexa says device not responding when name is spelled incorrectly. Always it's not the problem with the server 😃

amitabh-srivastav, 2019-01-02:
Since, the problem was for **all of my Sinric devices** and they were being shown on Alexa app (device settings) as "**Device is unresponsive**" , it could not have been spelling problem.

kakopappa, 2019-01-02:
> Any fix for this yet?

In an hour or so. Almost at home now

dhpete, 2019-01-02:
Lol. Sorry, man...take your time. Happy New Year.

kakopappa, 2019-01-02:
Hi guys,

This issue is related to ios alexa app skill linking issue fix I made a few days ago. When Alexa renews the oauth token, i have sent an invalid token back to alexa. Now when you ask Alexa to do something it's sending a request with an invalid token and server has no idea to which account is making the request 

unfortunately, to fix this it requires to link the re-link (disable/enable) the skill so it will send a valid token next time.