kakopappa/sinric #186: May I say that this software kinda sucks?

CollaVinilica, 2019-01-02:
The Skills doesn't always work, I had to disable and enable it already 3 times in two days to make it work for just a couple of hours, the devices stop responding after a a couple of hours and I have to reboot them. Too many times, heading to the website I had a cloudflare error of some kind. The developer doesn't replay to any help request regarding his own code. Should I just migrate to something else? Just le me know.

kakopappa, 2019-01-03:
1. There was a problem with a change I pushed to production on 31st December. It requires to relink the skill after I fixed the issue. I fixed it like 8 hours ago. for details https://github.com/kakopappa/sinric/issues/184

2. I don't have a LED strip but I tried to help you anyway then you said " I already know to how to use PWM in the code," If you know PWM you should know how to convert the brightness level to PWM.
@servermcp tried to help you as well. 

3. migrate to something else.  I am not going to stop you trust me

bangs6996, 2019-01-03:
I had  to reset my stuff as well yesterday and they are all down right now.. 
"sorry not responding"  or "I'm not quite sure what went wrong"
However I'm still very pleased with the server and just learnt how to send the current temperature to the server..so thats cool..     My only thing would be if you add something  to the  main page saying the servers are down or on repair or  to do as you said above..     regardless thanks for the Server and your time. keeping it updated..

kakopappa, 2019-01-03:
> Same issue here, "sorry ... not responding"
>
> Would you please consider making the code/skill/gui available ? We try to
> help

Have you tried linking the skill back? I fixed the issue few hours ago, you
have to relink the skill to make it work again

gizmocuz, 2019-01-03:
@kakopappa , best wishes for 2019 !!

Just disabled the skill in alexa, enabled it again and discovered new devices.

Working now!

Maybe you could find a way to notify all registered users by mail for these kind of changes....

Glad it's working again (and my wife thanks you as well !)

malhotrasujeet, 2019-11-04:
I keep getting 'Sorry, ........ is not responding' for random devices. this is happening very often from last one month. tried unlinking and linking the skill again, didn't help. also when i try to play a routine with multiple devices, it operates all devices but will leave one random device unchanged. anybody facing the same issue? Any leads on this @kakopappa ?

kakopappa, 2019-11-04:
@malhotrasujeet I will take a look at the server. I haven't looked at it for a while now. I recommed you to move to Pro (https://sinric.pro) (not officially public yet)

gizmocuz, 2019-11-04:
Experiencing the same issue. My wife knows to tell alaxa sometimes 3 times to turn of a light.....

@kakopappa , do we have to register a new account or should we be able to use the current login information from the other site ?

kakopappa, 2019-11-04:
@gizmocuz have to register again. sorry about that. 

This is a new system. I rewrote the whole thing again to scale horizontally. Teamed up with couple of guys to build SDKs for ESP/node/python as well. They did a outstanding job. We did this to avoid issues like flooding the server with requests and sending wrong device ids ect. plus we can suppor new sensors like Motion and Conact now with new code.. I haven't annouced it yet because there are few trivial issues to be addressed. Overall it's solid platform.

gizmocuz, 2019-11-04:
@kakopappa , It looks very great thank you ! Going to experiment with it soon.
I do noticed that the links to the example projects are not correct (quickstart, 4.3 Download the demo code from here), but i found them on your github repository

kakopappa, 2019-11-04:
@gizmocuz Thanks for letting me know. I will get it fixed

gizmocuz, 2019-11-04:
@kakopappa , is there a place we can have a chat about the pro version ? Do you have a Slack channel or some other ways to communicate ?
I receive a 'deprecated' message when i compile the example, like this line is not needed:

`SinricPro.add(mySwitch);`

as this is done above with

`SinricProSwitch& mySwitch = SinricPro.add<SinricProSwitch>(SWITCH_LIGHTS);`

The problem i am having is that the 'Alexa skill' finds the device, but i can not control it...
Clicking On/Off in the Alexa app says problem communicating with device
Nothing shows up in the device log

kakopappa, 2019-11-04:
@gizmocuz Do you mind raising an issue here with your code (without keys)?
https://github.com/sinricpro/esp8266-esp32-sdk with the code you used?

Boris @sivar2311 will take a look at it and respond. If the Alexa says "problem communicating with device" that means esp to server connection is not working

kakopappa, 2019-11-04:
@gizmocuz SinricPro.add(mySwitch); seems to be related to outdated documentation. 

Please change to use https://github.com/sinricpro/esp8266-esp32-sdk/tree/master/examples/switch

sivar2311, 2019-11-04:
In fact, there was a mistake in the Readme. This is fixed now.
If there are sill issues, please oppen issue in correct repo: https://github.com/sinricpro/esp8266-esp32-sdk (including code without credentials)
Thank you!

gizmocuz, 2019-11-05:
@kakopappa , thanks i will, but need to investigate more. I see i can successfully switch in the dashboard and it is received my the ESP. Only Alexa sees the device as not responding..

sivar2311, 2019-11-05:
@gizmocuz Please enable Debug Info on Serial monitor by putting this before #include "SinricPro.h":
```
#define DEBUG_ESP_PORT Serial
#define NODEBUG_WEBSOCKETS
#define NDEBUG
```

This enables Debug on Serial port.
Please post your debug output here

gizmocuz, 2019-11-05:
@sivar2311 , no debug logs when switching in Alexa, Alexa says device is unresponsive
Do get logs when switching fron SinRic dashboard.
So i can't post any logs....

sivar2311, 2019-11-05:
There must be something in serial log...even during connection. Please post this log here (comment out your credentials).
Is device shown online in Dashboard?
Please check your credentials carefully (APP-KEY, APP-SECRET).

kakopappa, 2019-11-05:
I have setup a telegram channel, @gizmocuz are you on telegram ?
https://t.me/sinricpro

gizmocuz, 2019-11-05:
@sivar2311 , maybe it is the language, but as mentioned i do receive debug logs, but none when controlling via Alexa
@kakopappa , great ! But I am not able to type anything , maybe i have 'viewer' rights

sivar2311, 2019-11-05:
Please let us support you.
But without any information (source code, debug logs) this will get very hard.
Please give us more information by sharing your logs and your source code.

gizmocuz, 2019-11-05:
@sivar2311 , lets continue on Telegram ?