kakopappa/sinric #188: Ambient temperature and Temperature Setpoint on GH

Lwiggy, 2019-01-03:
Hi, first of all Great job!

I´m having some trouble when I call  setTargetTemperatureOnServer, I send a value and both of the Ambient Temperature and the Temperature Setpoint change to the same value i sent on GH. 

Is there a way to only send the Ambient Temperature?
Is there a way to only send the Setpoint Temperature?

Regards.

kakopappa, 2019-01-04:
@Lwiggy Thanks

When I wrote the API, I was only thinking about Alexa. Alexa support only targetSetpoint. Thats why it's like that.
https://developer.amazon.com/docs/device-apis/alexa-thermostatcontroller.html

Since a few more people have requested this, I will update the server to support 

"Setpoint", "TemperatureAmbient", "HumidityAmbient"

https://developers.google.com/actions/smarthome/traits/temperaturesetting
 
I will let you know when it's done.

Lwiggy, 2019-01-04:
@kakopappa Thanks for the fast reply.

Looking forward for that improvement.

Regards.

emnik, 2019-01-04:
@kakopappa
Please implement the same functionality (ambient temperature) also for
Alexa as it is now supported as you can see in the related interfaces of
the page you mentioned before. See
https://developer.amazon.com/docs/device-apis/alexa-temperaturesensor.html

kakopappa, 2019-01-05:
Done. 

Here is an example i made with dht 11 sensor

https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example_with_dht_11.ino

Lwiggy, 2019-01-07:
Thank You! 
I just got home made the changes and it worked!