kakopappa/sinric #193: Enable skill for Mexico

aampudia, 2019-01-05:
Hi!!  Alexa and Echo devices were introduced to Mexico a couple months ago.  is there a way you could enable the Skill for Mexico??  i have a 3rd generation echo dot, and some ESP8266's that i'd like to test...

if you need a translation or anything i could help you out with that!  

thank you!

kakopappa, 2019-01-06:
I think it should be available for you because it shows Spanish (MX) in the
dropdown for languages

https://www.amazon.com/dp/B078RGYWQQ

What happens when you select Spanish MX? Does it redirect? Let me know what
happens I like to know too.

aampudia, 2019-01-06:
ooopps!! my bad!!  i was trying to enable it on "amazon.com" store, not on "amazon.com.mx" store,   it redirected me to amazon.com.mx and i was able to enable it....

closed 2019-01-06 by aampudia