kakopappa/sinric #194: no connection  NodeMCU with internet

unknownBob, 2019-01-06:
Hello guys,
i am using the switch example from kakopappa to connect the nodeMCU witch the internet. The only change I did, is adding MySSID WifiPassword and MyApiKey. But there is no connection. Do you guys have any idea? The adding variable I checked allready 1000 times. What else can be the mistake?

output:

Connecting to Wifi: Wodafone-t@..
.........................................................................................................................................

Bob

unknownBob, 2019-01-06:
There is also no mistake mentioned ...

Der Sketch verwendet 455400 Bytes (43%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 31252 Bytes (38%) des dynamischen Speichers, 50668 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
Uploading 459552 bytes from C:\Users\rgold\AppData\Local\Temp\arduino_build_586865/Amazon_Alexa-ESP8266.ino.bin to flash at 0x00000000
................................................................................ [ 17% ]
................................................................................ [ 35% ]
................................................................................ [ 53% ]
................................................................................ [ 71% ]
................................................................................ [ 89% ]
.................................................

unknownBob, 2019-01-06:
There is also no mistake mentioned.

Der Sketch verwendet 455400 Bytes (43%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 31252 Bytes (38%) des dynamischen Speichers, 50668 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
Uploading 459552 bytes from C:\fgdgdhdhdjdjdjdjdjdhkgjkd
................................................................................ [ 17% ]
................................................................................ [ 35% ]
................................................................................ [ 53% ]
................................................................................ [ 71% ]
................................................................................ [ 89% ]
.................................................

kakopappa, 2019-01-06:
Are you connecting to 5G? Try connecting to 2.4G

Also check the WiFi Channels

You can also start a hotspot from your mobile and try to connect to it
confirm

unknownBob, 2019-01-06:
With a hotspot from my mobile phone it is working!!!!!! NICE

Now I try it witch wifi
I used seperate SSID for 2,4 and 5G, but it dont work with 2,4.
I also changed the wifi channel and the Wi-Fi Protected Access without success.
Any other ideas?

kakopappa, 2019-01-06:
I remember I had the same problem long time ago. I go remember changing the
WiFi channels and trying again

When you scan for WiFi can you see it?

https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/scan-examples.html

unknownBob, 2019-01-06:
Thanks alot for the incredible fast answer!!

I tried it with changeing the channel again. With the scan example I can see my wifi:
6: Test, Ch:2 (-45dBm)

But a wifi connection is not possilbe.

twlizer, 2019-02-12:
I am having the same issue, No Wifi connection. I've connected with Feauxmosp with the same Node. I as well have checked the SSID and Password 1000xs.