kakopappa/sinric #196: smartlock in alexa need a PIN code to open door.. where is?

jesusado, 2019-01-06:


jesusado, 2019-01-06:
Hi
Alexa need a PIN code for unlock the door.. This skill do not have it.

kakopappa, 2019-01-06:
hmm.. 

https://developer.amazon.com/docs/device-apis/alexa-lockcontroller.html... it does not say anything about PIN code.

jesusado, 2019-01-06:
When you tell her to open the door, Alexa replies that she is not authorized to open smart devices. When you tell her to close the door she replies that there is no response from the server.
In other smartlock skills they give you an authorization PIN.

Thanks.

kakopappa, 2019-01-06:
According to this article: https://www.cnet.com/news/alexa-can-unlock-your-schlage-smart-lock-now/

"The feature is turned off by default, so you'll need to log into the Alexa app on your Android or iOS device and select the "unlock by voice" setting to turn it on. Once you do, the app will ask you to pick a four-digit PIN code, which Alexa will ask you for every time you ask her to open up. "

seems to be configured using in Alexa app. May be that's why it is not in the documentation

jesusado, 2019-01-06:
After selecting the "unlock by voice", the device indicates that there is no response from the server.
Do not let open.

Thanks

kakopappa, 2019-01-07:
I have no idea why.. 

can you use a switch for now? I will take a look when I get a chance