kakopappa/sinric #198: Connection limitation?

aarner42, 2019-01-07:
Greetings,
  I've been converting a house full of Fauxmos (Belkin Wemos emulation) switches built on a LOLIN D1 MINI (esp8266) platform + Relay + LED switch.  All was working well until I added the 8th or 9th device.  After that point, the connection to iot.sinric.com became sporadic.  I've currently got about 12 devices built - Am I getting myself blacklisted or bounced out due to excessive traffic?  The devices all connect to WIFI just fine but aren't able to reach iot.sinric.com, or if they do they almost immediately disconnect.  I believe I've taken care to avoid spamming the server with state updates, the debug output from the devices hasn't shown any such error anyways.

Additionally, if I switch off the circuit breakers to power down a few rooms, devices that weren't connecting will suddenly connect.  Seems like there's something behind the scenes impacting because it worked great up until I got 8 or so esp8266s connected.  

Please advise, am happy to share source code or anything else required.
Thanks,
-Aaron Dutenhoefer
aarner@gmail.com

kakopappa, 2019-01-07:
If you turn off all devices and connect only one device, does it work?

kakopappa, 2019-01-07:
Following IPs has been blocked so far

79.13.115.184;
172.69.194.42;
188.114.102.186;
156.198.37.180;

I think the limitation coming from your router

aarner42, 2019-01-07:
It does work in that situation, yes.   As I start switching rooms back on, invariably the last few devices powered up start having connection issues.

kakopappa, 2019-01-07:
I can turn on logging on the server and we can try to diagnose what's going on now if you are up for it

aarner42, 2019-01-07:
Didn't see your earlier post about blocked IPs.  As none of them are mine I'll keep investigating on my side and verify it isn't router connection enabled.  Just to confirm though, once I've enabled everything, I'll have probably 50+ devices - do you see any potential scaling issues?  

BTW, other than this glitch I'm loving SinRic.  Very good job.  I'll be sending something to help with AWS hosting fees.

aarner42, 2019-01-07:
I'm at the office right now and away from home so right now doesn't work.   I first want to verify it's not router/firewall related on my end.  (I've been a dev long enough to know the fault is usually mine before blaming anyone else.)  If that checks out I'll get in touch to work out a time we can diagnose.  I'm in CST time zone, or GMT-5.

kakopappa, 2019-01-07:
50+ devices? wow that is a lot.. are going to automate the whole building? 

right now it runs on a single nodejs instance and has done a pretty good job so far. With the current growth rate, I will have to add clustering support soon.

You are exactly 12 hours behind my time.. when you go back home, I will be waking up to go to work. so check and let me know.

kakopappa, 2019-01-08:
just my two cents.

Why not use a relay controller module with 8/16 relays put it in a box and mount the box to wall or a din-rail and control it using an one ESP module? Keep everything centralized so it is easy to manage downside is you have to rewire to go trough the box

aarner42, 2019-01-08:
That would have been my first choice - unfortunately it's a rented house
and I can't make the sort of re-wiring changes in the walls that would be
needed to support a simpler architecture like that.  The way it's wired now
all the lights are shared with sockets across rooms so cutting the power at
the breaker would turn whole parts of the home off, not just specific
lights. :-(

I believe I have figured out the issue that was affecting the
connectivity.  Using wireshark I was able to determine that it was the
actual entire wifi session being dropped, not just the webSocket link.  The
cause was a 250 msec loop doing an analogRead of a current sensor (The
switches are in a 3-way config and I need to check whether the light is
drawing power to know the state).  There's a known issue with esp8266 that
analogRead() in a tight loop can cause this, and I had included a call to
yield() per the docs to mitigate.  Apparently that's wasn't sufficient for
some of my esp8266 modules.  Changing the yield() call to a delay(5) did
the trick.  Thanks for your input tho, it was helpful in focusing my
debugging efforts.

Please let me know in the future if there's anything I can do to help out
with Sinric.  My primary expertise is in Java on AWS/Windows/Linux, but
I've got some Node.js exposure and always eager to learn.

-a