kakopappa/sinric #200: Heat Mode always showing cooling on GH

Lwiggy, 2019-01-07:
Hi @kakopappa i got the ambient and setpoint temperature working now, thank you! but I got an issue in GH, when I change the mode in the app to Heating or Off it shows "Updating, please wait..." for a few seconds and always goes back to Cooling. I do receive the command on my board but the mode on GH always shows Cooling.  I tried using setThermostatModeOnServer but no change is made on GH.

Do you have any idea what could be going wrong? (I tested in Alexa and it works fine)

This is the message I see:
[image]

kakopappa, 2019-01-11:
OK i will take a look at this. Are you using the new sketch and new method to send parameters. Old way is obsolete now

Lwiggy, 2019-01-11:
Yes, I'm using the latest sketch.

Lwiggy, 2019-07-16:
Hi @kakopappa, Im trying to anayze this issue from my side,
From GH i receive:{"deviceId":"5cxxxxxxxxxxxxxxxxxxxxxxx","action":"action.devices.commands.ThermostatSetMode","value":{"thermostatMode":"cool"}} when i ask GH to put the thermostat in cool mode.
but when i call the setThermostatModeOnServer this sends to a different format to Sinric server:
 {"deviceId":"5cxxxxxxxxxxxxxxxxxxxxxxx","action":"SetThermostatMode","value":"COOL"}

Is this format for alexa? or Sinric server translates this to GH?

I don´t know if i can make some changes on the setThermostatModeOnServer to deliver GH format?

thank you for your help.

Lwiggy

kakopappa, 2019-07-16:
Sinric server sends what it gets.

You should send back what is allowed in GH
https://developers.google.com/actions/smarthome/traits/temperaturesetting

Lwiggy, 2019-07-17:
Did you change something on the server? It started working fine a few minutes ago.