kakopappa/sinric #201: Automation support

mrmojtabaa, 2019-01-09:
hi @kakopappa 
is there any way to turn on/off  a switch at specific time ? 

if there is no way
i think we can do it . 

[IFTTT webhooks](https://ifttt.com/maker_webhooks) is a place that we can make web requests.
therefore we can call Sinric API's. 
but there is a problem with webhooks ! we can't add form-data to our request like this 

`curl --request POST \ --url https://sinric.com/api/v1/updateDeviceSettings \ --header 'cache-control: no-cache' \ --header 'content-type: multipart/form-data; boundary=----WebKitFormBoundary7MA4YWxkTrZu0gW' \ --form action=setPowerState \ --form value=OFF \ --form deviceId=xxx \ --form apikey=xxx
`

we can just set this values

> URL 
> Method (Post ,Get ...)
> Content Type (Json , ...)
> Body

if you write an API that we can add token and other values in URL or Body , we will have Automation Support 👍

kakopappa, 2019-01-09:
Ok

I have no idea how their platform work. I assume it does a webhook call
when it’s the time to execute a command

Can you tell me how did you set it up? I will add this release to 1.7

mrmojtabaa, 2019-01-09:
thank you very much.
after sign in, open [https://platform.ifttt.com/dashboard](https://platform.ifttt.com/dashboard) page 
you should see somthing like this 

[image]

in the top right of page click on new applet, opens new applet page.
threre you can say
IF , THIS( time is 12 ) , THEN , THAT( make api call request )  
or  
IF , THIS( is it sunset ) , THEN , THAT( send an email ) , ...

[image]