kakopappa/sinric #203: Activate window shutters using "Raise" and "Lower" keywords

bilalasd, 2019-01-10:
I know that Sinric currently supports window shutters using the "channel" command. 

But recently I saw Link Shades(https://www.linkshades.com/pages/amazon-alexa-integration) that provide Alexa integration and you can activate that using the "Raise" and "Lower" commands.

Is it possible to do something like this with Sinric?

There is a small video on the link posted above.

kakopappa, 2019-01-10:
Looks like they are using PrecentageController

https://developer.amazon.com/docs/device-apis/alexa-percentagecontroller.html

I have already implemented it

https://github.com/kakopappa/sinric/wiki/Supported-devices-and-their-capabilities

I think saying Rise or Lower doesn’t have any effect.  only the % you
mention at the end does everything

What happens when you ask Alexa to do the same when using Sinric? It should
work

bilalasd, 2019-01-10:
I have a setup at home. I'll try it tonight and let you know.
Thanks,

bilalasd, 2019-01-11:
Just tested it and it works! I get the following message
[WSc] get text: {"deviceId":"xxxxxxxx","action":"AdjustPercentage","value":{"percentageDelta":-25}}

I guess they just check if the value is negative or positive and just raise or lower the blinds all the way.

closed 2019-01-11 by bilalasd