kakopappa/sinric #204: Need help to make it work Neopixel

Tjupi, 2019-01-10:
i register at Sinric etc, all works and is connected..
problem is i can turn on and of the Neopixel, setting a pin etc in the code, im totally blank in programming, i defined a #define PIN 5, and in setup i have pinMode(PIN, OUTPUT);  Is this enough, becourse nothing happends at all, on the board ESP8266 only on TX pin i can turn on and off, but cant change colours set brightness etc..

kakopappa, 2019-01-11:
I do not have a Neopixel, so I am not sure how to help you with this. 

Can you make it work using ESP?  There are lot of turorials on youtube. If you can control using ESP then it's very easy to copy the code to sinric sketch

Aggebitter, 2019-03-09:
Got some ugly code here thats is using neopixelring: https://github.com/Aggebitter/KimBell/tree/master
Demo here: https://youtu.be/68NlSyKhukI