kakopappa/sinric #206: HELP needed for using API

RobertoPrincipato, 2019-01-14:
Good morning guys, 
I ask you for help in using API. Could someone suggest an example of login, I am not able to do it because of an error named 404, probably I will miss the request. 

Thanks for your excellent work and many compliments. 

Sorry for my English, I am still a beginner!

kakopappa, 2019-01-15:
Hi, 

1. Make sure your IP is not blocked. https://github.com/kakopappa/sinric/issues/207. 
If it is let me know.

2. API documentation is here: 
https://github.com/kakopappa/sinric/wiki/API-Documentation

RobertoPrincipato, 2019-01-15:
Thanks in advance for the pront reply. 
My IP address is not blocked, but unfortunately my level of knowledge of the subject is low, so please I ask you to write an example of an API request so i can modify it and understand it.
Waiting for your reply i wish you a good day

kakopappa, 2019-01-18:
You can use Postman
https://www.getpostman.com/downloads/

to test the queries in the API documentation and generate the code for it