kakopappa/sinric #210: Sinric is down, Error 525

Deimos1994, 2019-01-17:


amitabh-srivastav, 2019-01-17:
@kakopappa Host error. probably server is down.
Error 525 Ray ID: 49aaa202386469f5 • 2019-01-17 17:44:58 UTC
SSL handshake failed

malhotrasujeet, 2019-01-17:
Alexa devices gone unresponsive. 
Sinric skill not getting enabled. shows the same error as mentioned above by @amitabh-srivastav

aryeguetta, 2019-01-17:
Just wanted to learn the skill with Alexa and Google and saw that Error 525 SSL handshake failed.
Is that error fixable? thanks for a great skill. Hope it will be back soon.

CptZeroG, 2019-01-17:
Auch! Same problem...half of my house is down...I hope that will work back soon

GaryLittlemore, 2019-01-17:
My devices are down too :-(

acostadaniel1, 2019-01-17:
Well.. never works as expected for long time, but reconnecting with Google home, it was working for a few days..
But now the server is down :(

kakopappa, 2019-01-18:
thanks for letting me know. I have fixed the issue and all services should be back to normal now.

The reason was I had a cron job to renew the server certificate in the midnight. For some reason, certbot (letsencrypt) failed today and it did not release port 443 for the Nginx server after it crashed. So no one was listening on port 443. thus

Error 525 Ray ID: 49acc4514db181b6 • 2019-01-17 23:57:55 UTC
SSL handshake failed

Deimos1994, 2019-01-18:
Works now, thanks @kakopappa, closing.

closed 2019-01-18 by Deimos1994