kakopappa/sinric #211: Frequent Disconnection Isuues

rituraj841, 2019-01-17:
Getting these disconnection issues in Serial Output for NodeMCU.

00:02:21.563 -> [WSc] Webservice disconnected from sinric.com!
00:02:26.912 -> [WSc] Webservice disconnected from sinric.com!
00:02:32.378 -> [WSc] Webservice disconnected from sinric.com!
00:02:37.824 -> [WSc] Webservice disconnected from sinric.com!
00:02:43.262 -> [WSc] Webservice disconnected from sinric.com!
00:02:49.006 -> [WSc] Webservice disconnected from sinric.com!
00:02:54.436 -> [WSc] Webservice disconnected from sinric.com!
00:02:59.883 -> [WSc] Webservice disconnected from sinric.com!
00:03:05.323 -> [WSc] Webservice disconnected from sinric.com!
00:03:10.791 -> [WSc] Webservice disconnected from sinric.com!
00:03:16.356 -> [WSc] Webservice disconnected from sinric.com!
00:03:21.810 -> [WSc] Webservice disconnected from sinric.com!
00:03:27.288 -> [WSc] Webservice disconnected from sinric.com!
00:03:32.907 -> [WSc] Webservice disconnected from sinric.com!
00:03:38.379 -> [WSc] Webservice disconnected from sinric.com!
00:03:44.092 -> [WSc] Webservice disconnected from sinric.com!
00:03:49.505 -> [WSc] Webservice disconnected from sinric.com!
00:03:54.965 -> [WSc] Webservice disconnected from sinric.com!
00:04:01.041 -> [WSc] Webservice disconnected from sinric.com!
00:04:06.071 -> [WSc] Webservice disconnected from sinric.com!
00:04:11.497 -> [WSc] Webservice disconnected from sinric.com!
00:04:17.021 -> [WSc] Webservice disconnected from sinric.com!
00:04:22.695 -> [WSc] Webservice disconnected from sinric.com!
00:04:28.097 -> [WSc] Webservice disconnected from sinric.com!
00:04:33.556 -> [WSc] Webservice disconnected from sinric.com!
00:04:39.067 -> [WSc] Webservice disconnected from sinric.com!
00:04:44.505 -> [WSc] Webservice disconnected from sinric.com!
00:04:49.941 -> [WSc] Webservice disconnected from sinric.com!

rituraj841, 2019-01-17:
@kakopappa

aeronixil, 2019-01-17:
check the sinric.com,it has been offline since an hour

JoaquimLopes1980, 2019-01-17:
Yes. Same problem. Maybe in maintenance right?

rituraj841, 2019-01-17:
Maybe..Maintenance stuff going on [Sinric](https://sinric.com/)

saikhurana, 2019-01-17:
maybe maintenance but it's past midnight where Aruna lives so ,i don't know let's hope the server is up soon

saikhurana, 2019-01-17:
irdevkit website is also down

JoaquimLopes1980, 2019-01-17:
Maybe need some support or money, a donation maybe...

saikhurana, 2019-01-17:
i would love give irdevkit a try soon enough. saving for that. sinric is such an amazing skill. he deserves every bit of donation

kakopappa, 2019-01-18:
thanks for letting me know. I have fixed the issue and all services should be back to normal.

The reason was had a cron job to renew the server certificate in the midnight. Cert bot failed and it did not release port 443 for nginx start the service.

dhpete, 2019-01-18:
Thanks for all you do! Youre the best!!!