kakopappa/sinric #213: All my google home can not reach Sinric

aryeguetta, 2019-01-20:
Hello, I have worked with Sinric on my Google Home mini (two devices) and for an unknown reason it stops working - Every time it say "Sorry, I could not reach Sinric" when I ask to sync - It say that it got an Error while connecting to Sinric, I would be happy to know if that is on my End or Sinric's End.
I am not sure what I should do now?
A couple of days ago we had Error with both Alexa and Google Home - It seems like Alexa is fixed now BUT for Google Sinric is unreachable - Please assist.
Thanks

kakopappa, 2019-01-20:
Let me check the server status

kakopappa, 2019-01-20:
No errors on the server. Seems to work fine. here is proof

[image]

try disabling and enabling the skill again.

shreyamt, 2019-01-22:
I faced the same issue and I know the reason: in the Google Home guide, it says you have to "TEST" the skill. The problem is, "TEST" services only last for 30 days on the Google Actions console, after which it stops working. Then I have to unlink the sinroc account from Google, re-deploy the service as a test and then relink the account. Also, I have to set up all the routines again. Is there any way to deploy the service permanently, so this exercise is not required after every 30 days?

kakopappa, 2019-01-23:
@aryeguetta Last time I checked, They asked me to ship two devices for them to test the hardware and approve it if I want the action in the store.

servermcp, 2019-01-24:
Can WE($) do that?

kakopappa, 2019-01-24:
@servermcp bribe an google employee?

servermcp, 2019-01-24:
Sending you money to print the pcb, buy components and print in 3D a case of, for example, a smart socket. There are many people interested in Sinric with knowledge of electronics that could help with the design of the pcb and those who do not, we collaborate with money! let sinric fly!!

Deimos1994, 2019-01-24:
@kakopappa, I just curious what have openHAB shipped to Google? as far as I understand they don't make hardware... and there are other projects that aimed at DIYers that get approved, why not Sinric?
What hardware does Google want? a couple of esp8266's?
Maybe you should explain to them that Sinric doesn't make hardware, and that the whole point is to make it yourself.
By the way, I've used python to connect my PC to Sinric (just for testing), and it works, so you can try to offer them a python code instead, so they can test Sinric without any additional hardware.

Fleker, 2019-01-25:
In order to verify that actions meet policy guidelines, there is a review process. In the domain of smart home, which is based around physical devices, this review process requires a physical certification process. This may include hardware manufactured by a company or whatever hardware and software setup is required for a particular service.

I understand this may be additional overhead for these maker-focused services, and I would be willing to help get to an MVP that can be certified. However, this requirement isn't going to be waved. I can try to see how similar groups such as openHAB did their certification, which could help guide this project forward.

kakopappa, 2019-01-26:
Ok so, linking error "401. That’s an error. Your client does not have permission to the requested URL blah blah. That’s all we know. " seem to be related to iOS Safari. Clearing history and website data from iOS Home-> Settings > Safari seems to fix it.  I have updated the guide accordingly.

Next, we have to get the action approved. Let's give some time to Nick to see what he comes up with

Fleker, 2019-02-06:
All smart home actions need to go through physical certification. This can be shipping a Raspberry Pi with the software to the facility for review, which is what similar DIY platforms have done.

qwandor, 2019-02-28:
Any news on getting the Google Home integration certified so we don't have to each set up a test account and re-enable it every 30 days? I'm also happy to contribute towards a fund for sending a test device to Google, or even put together an example device with a few LEDs and some decent firmware and send it myself, if that would help.

kakopappa, 2019-03-01:
Anyone here from US? my problem is shippnig to USA testing ceneter costs me like 60-80$ for a 7$ item which I really do not like... I can buy a ESP module with relay controller and ship it  someone in US is willing to help me out here.

al8943, 2019-03-01:
@kakopappa, I live in Colorado USA, I would happy to help.

al8943, 2019-03-01:
I can send this in your behalf, you would just need to send me the information. I am happy to donate it for the project.
  https://www.ebay.com/itm/Mini-Size-ESP8266-Relay-Module-10A-220V-Network-Relay-WIFI-Module-Board-ZP/362546557795?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

I also have a design of my own that I use to drive an RGB strip and water pump on a fountain that I can send in.

kakopappa, 2019-03-02:
That’s awesome. Thank you.

I can pay for this and ship this to your address if you wish. Do you have
access to a 3d printer? I was thinking may be we need to print a case for
this before sending it so it doesn’t look weird. What do you think?

I have created a channel in Slack #sinric if anyone interested in joining.

Cheers

al8943, 2019-03-02:
There is no need to send anything, I have a 3d printer and I think I can
put something togherther in the next week, then I wil send you pictures to
see if OK with you.
Cheers
Al.

kakopappa, 2019-03-04:
Thanks. 

I have resubmitted the google home action again for approval. 

They should get back to me with in cople of days from now. I clearly said Sinric does not manufature hardware and if required we can ship one. Let's see their reponse.

Thanks all

kakopappa, 2019-03-07:
Here is the progress so far: I will update it as it progress

Project submission - **Done**
Policy review Initiated - **Done**
Submission of Compatible Device Form - **Done**
Policy review passed and email confirmation is sent - Waiting
Completion of Intake Form (to send devices for testing) - Pending
Devices sent for testing and certification (instructions in Policy Email) - Pending
Devices pass testing and your project launches! - Pending

al8943, 2019-03-07:
If necessary i have this ready to ship.
It incorporates  https://github.com/the-real-orca/WiFiManager, OTA updates
and responds to all google assistant and alexa light actions.
If you want me to send it, I would need to know what you for DeviceID,
APIKey and name for AP SSID and OTA update hostname.
I don't know if I can send the pictures this way, if you don't receive them
give me an email where I can send them.
Al.

kakopappa, 2019-03-07:
Please drop me an email

aruna.tennakoon@gmail.com

I will send you the API Key and Sinric account login for GH testing

Let me know if you need any help with testing.

servermcp, 2019-03-07:
Kakopappa, can we make a permanent thread on this topic? To follow it closely

tonyshree18, 2020-12-07:
For me the Switch is showing in google app but the relay is not turning on. But the same time I checked with alexa it worked. But not with google assistant. I used a different code in arduino and I am using ESP32. Help me

kakopappa, 2020-12-07:
@tonyshree18 On/Off command format is different in Google Home. Check the google home example code

tonyshree18, 2020-12-07:
> No errors on the server. Seems to work fine. here is proof
> 
> [image]
> 
> try disabling and enabling the skill again.

I too received the screen like you. But the relay is not switching on. I tried the same with alexa it worked but no with google assistant

kakopappa, 2020-12-07:
Have you changed your code to support google home?

Notice the difference in action?

https://github.com/kakopappa/sinric/blob/master/arduino_examples/google_home_switch_example.ino

Line 91

tonyshree18, 2020-12-07:
Then what are the output pin numbers to connect with my relay module??

tonyshree18, 2020-12-08:
Hello Sir,

            Currently, I am using this code to run my project :

#include <SocketIOclient.h>
#include <WebSockets.h>
#include <WebSockets4WebServer.h>
#include <WebSocketsClient.h>
#include <WebSocketsServer.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
#include <libb64cencode.h>

#include <Arduino.h>
#include <WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //
https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> //
https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;
WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "MYAPIKEY" // TODO: Change to your sinric API Key. Your
API Key is displayed on sinric.com dashboard
#define MySSID "SSID" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "PASSWORD" // TODO: Change to your Wifi network
password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// Switch
const int BUTTON1_PIN = 32;
const int BUTTON2_PIN = 35;
const int BUTTON3_PIN = 34;
const int BUTTON4_PIN = 39;

//Relays
const int RELAY1_PIN = 15;
const int RELAY2_PIN = 2;
const int RELAY3_PIN = 4;
const int RELAY4_PIN = 22;

//Status LEDs
const int LED1 = 26;
const int LED2 = 25;
const int LED3 = 27;

String device_ID_1 = "DEVICEID";
String device_ID_2 = "DEVICEID";
String device_ID_3 = "DEVICEID";
String device_ID_4 = "DEVICEID";

ButtonConfig config1;
AceButton button1(&config1);
ButtonConfig config2;
AceButton button2(&config2);
[... 321 lines cut ...]
buttonState) {
  Serial.println("EVENT4");
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer(device_ID_4, "ON");
      digitalWrite(RELAY4_PIN, LOW);
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      setPowerStateOnServer(device_ID_4, "OFF");
      digitalWrite(RELAY4_PIN, HIGH);
      break;
  }
}

This code is perfectly running with my Alexa, but not with Google
Assistant. You have shared me a screenshot on which the light works. The
same screen appeared to me. No errors in Google Assistant, I have
successfully linked my sinric account and added my devices. Then I tested
to turn on the relay. It shows that the relay is turned on, but the relay
didn't. But, With Alexa the relay turned on. I am using ESP32 WiFi
Development Board. The pin of First relay is 15 and second is 2, third is 4
and fourth is 22. Please help me connecting. In your code what are the pin
numbers that I needed to connect? Please tell me

kakopappa, 2020-12-08:
@tonyshree18  

Google Home / Alexa has a different command format. first test using this Google Home sketch whether Google Home works.
https://github.com/kakopappa/sinric/blob/master/arduino_examples/google_home_switch_example.ino#L91

Second.
Change this line if (action == "setPowerState") to if (action == "setPowerState" || action == "action.devices.commands.OnOff")

tonyshree18, 2020-12-08:
Okay I asked you what are the pin numbers writen in the code?? To connect
with my relay module?

tonyshree18, 2020-12-08:
This is the error message I got from Arduino IDE while compiling :

Arduino: 1.8.13 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600,
None"

In file included from
C:\Users\tonys\OneDrive\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:33:0,

                 from
C:\Users\tonys\OneDrive\Desktop\Google_Assistant_Home_Automation_using_Sinric\Google_Assistant_Home_Automation_using_Sinric.ino:6:

C:\Users\tonys\OneDrive\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiType.h:26:19:
fatal error: queue.h: No such file or directory

compilation terminated.

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Please help me, I am getting many errors

kakopappa, 2020-12-08:
@tonyshree18 Sketch is written for Esp8266, you are compiling it on Esp32. 

If you have a further question open another issue

closed 2020-12-08 by kakopappa

tonyshree18, 2020-12-08:
Hi, I uploaded the sketch. What are the pins that I need to connect. please tell me. I don't find any pin numbers in the sketch