kakopappa/sinric #214: Speaker becomes an outlet and a suggestion/wish

kent2019, 2019-01-23:
Hi,

Thanks for the very nice “sinric” service !

I tried it for the first time yesterday and noticed:

When I created a speaker device on SINRIC, it shows up as an outlet in “google home”.
Do you plan to make a more general google home integration so it is possible to receive text/ingredient to our devices ?

I need something like:
“hey google Berta Turn on office light” or “hey google Berta Turn off mp3 music” (in Danish)

Where “Berta” I like a second trigger word and “turn on office light””turn off mp3 music” is the text I like to transfer from google -> senric -> our device.

(I know IFTTT can do it – but it doesn’t support Danish)

Best regards
Kent

kakopappa, 2019-01-23:
hi @kent2019 problem is Google does not support speaker yet. Same as lock. #178 

closes thing i found closer to speaker is outlet 
https://developers.google.com/actions/smarthome/guides/

kent2019, 2019-01-23:
Thanks for the quick response, please let me know I you plan make something like: 

“hey google Berta Turn on office light” or “hey google Berta Turn off mp3 music” (in Danish)
Where “Berta” I like at second trigger word and “turn on office light””turn off mp3 music” is the text I like to transfer from google -> senric -> our device.

(Or you know that will not gonna happent)

kakopappa, 2019-01-23:
second trigger keywords are supported by Google. Normally it's hey google Turn on office light

kent2019, 2019-01-23:
The "berta" trigger word is just to let google know the voice command is ment for  "sinric" - I would like to receive the fuld text after "barta" on my esp8266 ! (I want to use Goggle home as little as possible and do the text analysis my self.)