kakopappa/sinric #215: Could not convert 'temperature' from 'float' to 'String' on DHT11 example

frappesc, 2019-01-24:
https://github.com/kakopappa/sinric/blob/a5c4dcf61513cbe1b03188512ae7ebfcd92e3cf3/arduino_examples/thermostat_example_with_dht_11.ino#L187

when I try to compile this error appears to me. Is it my problem or is there something wrong with the code? Thank you
(I replaced deviceId with the id of the dht11)

Deimos1994, 2019-01-24:
@frappesc 
Make sure that you did not accidentally modified your code
https://github.com/kakopappa/sinric/blob/a5c4dcf61513cbe1b03188512ae7ebfcd92e3cf3/arduino_examples/thermostat_example_with_dht_11.ino#L212
Is it `float setPoint` in your code as well? maybe you accidentally modified it to `String setPoint` ?

post your sketch

frappesc, 2019-01-24:
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>
#include "DHTesp.h" // https://github.com/beegee-tokyo/DHTesp

#define DEBUG_WEBSOCKETS true

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
DHTesp dht;

#define MyApiKey "d618c631-a17a-4791-a496-ecf95b1cd3e8" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "xxx" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "xxx" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 500000 // 5 Minutes 
#define TEMPRATURE_INTERVAL 300000 // 3 Minutes

#define SERVER_URL "iot.sinric.com" //"iot.sinric.com"  
#define SERVER_PORT 80 // 80

uint64_t heartbeatTimestamp = 0;
uint64_t tempratureUpdateTimestamp = 0;
bool isConnected = false;

int device1 = 5;
int device2 = 4;
int device3 = 0;

void setPowerStateOnServer(String deviceId, String value);
void setSetTemperatureSettingOnServer(String deviceId, String value, String scale);
void readTempature();

void turnOn(String deviceId) {
  if (deviceId == "xxx") // Device ID of first device  (scrivania, cavi rossi)
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device1, LOW);
  }
  else if (deviceId == "xxx") // Device ID of second device  (principale, cavi gialli)
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device2, LOW);
  }
  else if (deviceId == "xxx") // Device ID of third device  (armadio, cavi marroni)
  {
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(device3, LOW);
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);

  }
}

void turnOff(String deviceId) {
  if (deviceId == "xxx") // Device ID of first device   (scrivania, cavi rossi)
  {
    Serial.print("Turn off Device ID: ");
    Serial.println(deviceId);
    digitalWrite(device1, HIGH);
  }
  else if (deviceId == "xxx") // Device ID of second device  (principale, cavi gialli)
[... 287 lines cut ...]

  webSocket.sendTXT(databuf);

}

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server.

void setThermostatModeOnServer(String deviceId, String thermostatMode) {

  DynamicJsonBuffer jsonBuffer;

  JsonObject& root = jsonBuffer.createObject();
  root["deviceId"] = deviceId;

  root["action"] = "SetThermostatMode";

  root["value"] = thermostatMode;

  StreamString databuf;

  root.printTo(databuf);

  webSocket.sendTXT(databuf);

}

frappesc, 2019-01-24:
is basically the dht11 example with my current code, just for testing (if there's something wrong I'm sorry, I'm not good at all, I'm just trying lol)

kakopappa, 2019-01-25:
try this.
https://gist.github.com/kakopappa/d69a356f2c413d698e574c9785ad75f9

I think i forgot to change the fuction signature definition on top. stange my IDE didn't warn me about this

Deimos1994, 2019-01-25:
@frappesc, just tried your code, It compiled successfully on my PC, I don't see any errors.

`Sketch uses 354200 bytes (33%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33224 bytes (40%) of dynamic memory, leaving 48696 bytes for local variables. Maximum is 81920 bytes.`

well, you can try to reinstall Arduino IDE, that may help.

frappesc, 2019-01-25:
ok now it worked. is the code i posted ok to work with google home? because when I ask how many degrees there are google home tells me that the thermostat is set to 0 degrees and at this time can not communicate with the thermostat.
Thanks in advance

Deimos1994, 2019-01-25:
I don't use DTH sensor, so maybe I don't understand something... 
`dht.getTemperature();` returns celsius correct? so why we send it to Sinric as second argument?
I think this line:
`setSetTemperatureSettingOnServer("<device id>", temperature, "FAHRENHEIT", temperaturefh, humidity);`
should be like this:
`setSetTemperatureSettingOnServer("<device id>", temperaturefh, "FAHRENHEIT", temperaturefh, humidity);`

That is what I did in my code, and it works fine, just try it @frappesc, it may solve the problem.

frappesc, 2019-01-27:
i'll try asap. i don't need the fahrenheit so i Will try sending  : 
setSetTemperatureSettingOnServer("<device id>", temperature, "CELSIUS", temperature, humidity);

frappesc, 2019-01-27:
finally, GHome tells me the temperature. Instead the humidity does not report it in the Ghome nor Alexa app. Also lately I'm having disconnections but on this I'm working on it, it could be my problem. Sorry if I keep getting problems, I really thank you