kakopappa/sinric #216: smartlock in alexa not responsive

RUBIKOF, 2019-01-25:
I have a problem and that is when I want to open the door after saying the key alexa tells me that my device does not respond (the door), with the lights I have no problem, likewise I do not know if I am using it better way

void turnOn (String deviceId) {
  if (deviceId == "xxxxxxxxxx")
  {
    Serial.print ("Turn on device id:");
    Serial.println (deviceId);
    digitalWrite (D1, HIGH);
  }
}
void turnOff (String deviceId) {
   if (deviceId == "xxxxxxxxxx")
   {
     Serial.print ("Turn off Device ID:");
     Serial.println (deviceId);
     digitalWrite (D1, LOW);
   }
}

this is what I'm using to open the door, it's the same code I used to turn on the lights.
Is it okay to use it like that? or what is the correct way to manipulate the door option?

RUBIKOF, 2019-01-28:
> 1. does it say "device does not respond" or "your device is not supported"? If the lights are working, lock should work too
> 2. Code looks fine. I assume you have  `pinMode(D1, OUTPUT); `  in setup()

in fact, the lights work wonders, turn on and off at my whim, but when I give the instruction to alexa to unlock the door then alexa responds by asking for the password, after giving the password alexa responds that he will unlock but after a few seconds he tells me that "Door" does not respond "and never unlocks the door.
I am using NodeMCU 1.0 (ESP-2E)

tomy983, 2020-01-02:
I am also trying to get the door to unlock. I give the command to alexa, she asks for the password and then nothing. no response, and no command to the esp. I have also lights working, with brightness and they all work, also reporting back their state to the server. The lock is configured exactly as a light bulb.

Should these parameters ( action == "setPowerState" and value == "ON" still ) also work for the lock or should they be different?