kakopappa/sinric #218: Smart lock

teohell, 2019-01-29:
Hi,
would it be possible to have an example on how to control a smart lock?
Many thanks!
Matteo

kakopappa, 2019-01-30:
Few users have reported that there is a problem with the lock controller asking for a PIN. I have not figured that part yet. (I think it requires to disable security sesstings. ) However, you can use a switch sketch to turn on/off.

teohell, 2019-01-30:
Hi, many thanks for your kind reply
I'd like to try creating lock device but I cannot find reference for how the payload is constructed for a lock device. If you mind referencing the type of action in the payload, I'll give it a shot!

strange-07, 2020-12-30:
> Hi, many thanks for your kind reply
> I'd like to try creating lock device but I cannot find reference for how the payload is constructed for a lock device. If you mind referencing the type of action in the payload, I'll give it a shot!

@teohell Any luck in finding " the type of action in the payload"?

teohell, 2020-12-30:
Unfortunately not
I'm currently using a switch device and then I handle everything else with Alexa routines

Inviato da iPhone

strange-07, 2020-12-30:
> Unfortunately not I'm currently using a switch device and then I handle everything else with Alexa routines Inviato da iPhone
> […](#)

@teohell Okay Thakyou...