kakopappa/sinric #225: add buttons for swirchs and lights

osama-Akt, 2019-02-09:
Hi , i wanna thank you about this open source iot, 
i'm running a nodemcu for like a month or more using the switch example sketch controlling 5 lights and 2 switches, it's working flawlessly with the app and with alexa ,
but i want to attach 5 buttons in nodemcu side (buttonState = digitalRead(buttonPin) or analogread(to save more pins); the button activates the relay so the light turns ON, but i cant figure it out how to send the state of that light turned on manually buck to the Sinric app or Alexa App ?
and i noticed that the nodemcu emitted an AP SSID under the name of ESP-xxxx (for digits) open network connection !!!!, i cant find in the sketch where is the command responsible of that  AP !
thanks again and keep up the sharing (y)

kakopappa, 2019-02-12:
hi, 

Thanks.. To send the status to server use this sketch.

https://github.com/kakopappa/sinric/blob/master/arduino_examples/how_to_update_server_example.ino

You can see the setPowerStateOnServer at the bottom. Call this ONCE when the status change.  Please send it once when it change the state. 

Do not send it in the loop(). 

> i noticed that the nodemcu emitted an AP SSID under the name of ESP-xxxxxx (6 char) open network connection !!!!, i cant find in the sketch where is the command responsible of that AP !

This is not my code... take a look at WIFI_AP, WIFI_STA, WIFI_AP_STA modes

philb1192, 2019-04-11:
> Hi , i wanna thank you about this open source iot,
> i'm running a nodemcu for like a month or more using the switch example sketch controlling 5 lights and 2 switches, it's working flawlessly with the app and with alexa ,
> but i want to attach 5 buttons in nodemcu side (buttonState = digitalRead(buttonPin) or analogread(to save more pins); the button activates the relay so the light turns ON, but i cant figure it out how to send the state of that light turned on manually buck to the Sinric app or Alexa App ?
> and i noticed that the nodemcu emitted an AP SSID under the name of ESP-xxxxxx (6 char) open network connection !!!!, i cant find in the sketch where is the command responsible of that AP !
> thanks again and keep up the sharing (y)

Hi, did you have the code running? I'm trying to build a 3 way using Alexa and a physical switch. I Need a way to change the status of the server with the physical switch so that Alexa will know if the light is on or off. I'm new in coding and i don't know where to start :/