kakopappa/sinric #226: Turn on light by code

CptZeroG, 2019-02-11:
Hi y'all,
I would to know if is possible to call an action, for example light up a LED (registered on SINRIC), from another board like NodeMCU.

The scenario is:
1) nodeMCU registered under SINRIC (connected via Skill to ALEXA), with a LED strip.
2) another nodeMCU with a button.

I'd like to turn on the LED strip using SINRIC (and, obviusly, then change the LED status),by pressing on the button of the second NMCU.

Can someone help me? Aruna?

kakopappa, 2019-02-11:
Send a UDP command from one to another?

CptZeroG, 2019-02-11:
Do you mean without pass trough the SINRIC server, send and receive command from and to NMCU and then change the state on SINRIC server? 
mhmm ... Maybe it's a solution, but works only if all the devices is under the same network.

It's no possible to send commands to SINRIC as, i suppose, the ALEXA server does?

kakopappa, 2019-02-11:
sinric does not support this. it just  relay commands from alexa to your esp module. look for a public mqtt server

CptZeroG, 2019-02-11:
Ok, all clear.
UDP connections or MQTT server.

Thanks Aruna.

closed 2019-02-11 by CptZeroG