kakopappa/sinric #227: Sinric on nodemcu

steve8266, 2019-02-12:
Hi, first github post hope in right place. Have loaded the github Sinric switch example onto nodemcu and it works fine. Can turn on and off via Alexa device (switch). But Alexa still allows on/off even when nodemcu is turned off, I.e should be unresponsive. What am I doing wrong please?

kakopappa, 2019-02-12:
hi @steve8266 welcome to Sinric world. 

This is a work-around I have implemented to get the Amazon Skill passed (Japanese) once it's done I will fix this back.

steve8266, 2019-02-12:
Many thanks. Sorry but I don’t fully understand. Are you saying you can provide a fix in near future?? Regards, Steve

nilava, 2019-03-02:
He meant that when skills are submitted for checking in alexa skill store, devices must be online for the test team to check, so he changed the if the devices can send its state back to server to off, so every device is online all the time!

steve8266, 2019-03-02:
Ah, now I understand. Thanks Nilava. Actually I've found the sinric devices rock solid thus far anyway and much more stable that the faux stuff I've tried in the past. So the lack of 'device can send state back' is not the big issue I thought it would be when I made the initial comment. Regards, Steve

closed 2019-03-02 by steve8266