kakopappa/sinric #229: how to change alexa response word okay

rashedomar, 2019-02-25:
how to change alexa response word okay

nilava, 2019-03-02:
It can't be changed, there is no option for custom response for Smart Home Skills. You could enable minimal response in alexa settings though, where it would just make a sound instead of saying okay.

temocastro, 2019-05-13:
Yes you can, create a routine, 

1.-Alexa, "your instruction" ("make me coffee")
2.-set the action (turn on relay 1)
3.-Alexa response "Your response" ("yes master, your coffee will be ready in 5 minutes")

send me an email if you need more help

rashedomar, 2019-05-13:
Can you register a video tutorial for this please?!

temocastro, 2019-05-13:
Please let me know if this helps, I just found this already in Youtube

Trigger your routine and define the response you want from Alexa after the trigger

temocastro, 2019-05-14:
https://www.youtube.com/watch?v=CO4l_nmjoEY

temocastro, 2019-05-25:
did you make it work @rashedomar ?