kakopappa/sinric #230: How can I add 8 relay?

canuri, 2019-03-04:
Hi how can i add more device?
I saw only two device in example code.
`void turnOn(String deviceId) {
  if (deviceId == "5c7282***************084") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(2, HIGH);
  } 
  else if (deviceId == "5c728b6b************42420") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(3, HIGH);
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);
    
  }     
}`

kakopappa, 2019-03-05:
you can extend the if condition like this

`
void turnOn(String deviceId) {
	if (deviceId == "5c7282***************084") // Device ID of first device
	{
		Serial.print("Turn on device id: ");
		Serial.println(deviceId);
		digitalWrite(2, HIGH);
	}
	else if (deviceId == "5c728b6b************42420") // Device ID of second device
	{
		Serial.print("Turn on device id: ");
		Serial.println(deviceId);
		digitalWrite(3, HIGH);
	}
	else if (deviceId == "xxxxx") // Device ID of second device
	{
		Serial.print("Turn on device id: ");
		Serial.println(deviceId);
		
	}
	else if (deviceId == "xxxxx") // Device ID of second device
	{
		Serial.print("Turn on device id: ");
		Serial.println(deviceId);
		
	}
	else if (deviceId == "xxxxx") // Device ID of second device
	{
		Serial.print("Turn on device id: ");
		Serial.println(deviceId);
		
	}
	else if (deviceId == "xxxxx") // Device ID of second device
	{
		Serial.print("Turn on device id: ");
		Serial.println(deviceId);
		
	}
	else {
		Serial.print("Turn on for unknown device id: ");
		Serial.println(deviceId);

	}
}
`

canuri, 2019-03-05:
> you can extend the if condition like this
>
> `
> void turnOn(String deviceId) {
> if (deviceId == "5c7282***************084") // Device ID of first device
> {
> Serial.print("Turn on device id: "); [...]

thank you

canuri, 2019-03-05:
my problem solved

closed 2019-03-05 by canuri