kakopappa/sinric #231: The relay have no trigger (nodemcu, Google Home)

ivan978, 2019-03-05:
I write the code for 6 relay but the Serial monitor say i have already trigger but the relay no trigger wut is the prob?

ivan978, 2019-03-05:
Maybe you want look my code??

kakopappa, 2019-03-05:
likely pins are not correct or not set

ivan978, 2019-03-05:
i have set all

ivan978, 2019-03-05:
i post the code here
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "xxxxxx" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "xxx" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "xxx" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

const int light1 = 5; // TODO: Change according to your board
const int light2 = 2;
const int light3 = 4;
const int light4 = 0;
const int AC = 12;
const int AC1 = 14;
const int FAN = 16;// TODO: Change according to your board

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;
 
#define DEVICE1 "xx"  //AC//TODO: Device ID of first device
#define DEVICE2 "xx"  //Fan
#define DEVICE3 "xx"  //light1
#define DEVICE4 "xx"  //light2  
#define DEVICE5 "xx"  //light3
#define DEVICE6 "xx"  //light4
//#define DEVICE7 "xxxxx"//TODO: Device ID of second device

void setPowerStateOnServer(String deviceId, String value);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == DEVICE1)
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    digitalWrite(AC1, HIGH);
     digitalWrite(AC, HIGH); // turn on relay with voltage HIGH
  }
  else	if (deviceId == DEVICE2)
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(FAN, HIGH); // turn on relay with voltage HIGH
  }  
  else  if (deviceId == DEVICE3)
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(light1, HIGH); // turn on relay with voltage HIGH
  }  
  else  if (deviceId == DEVICE4)
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    
     digitalWrite(light2, HIGH); // turn on relay with voltage HIGH
  }  
[... 159 lines cut ...]
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

// If you are going to use a push button to on/off the switch manually, use this function to update the status on the server
// so it will reflect on Alexa app.
// eg: setPowerStateOnServer("deviceid", "ON")

// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server. 
void setPowerStateOnServer(String deviceId, String value) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["deviceId"] = deviceId;
  root["action"] = "setPowerState";
  root["value"] = value;
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

ivan978, 2019-03-05:
i set the pin is to gpio 
or need to set like D1,D2

kakopappa, 2019-03-05:
code looks correct..

try turning on/off using a simple sketch without sinric

ivan978, 2019-03-05:
Ok i try

ivan978, 2019-03-05:
can run

CptZeroG, 2019-03-07:
Have you tried to a simple sketch and the relay works?
If not, maybe the problem would be the type of relay.
Some relay, using a nodemcu board, don't have enough current that they need from the pin.
In this case the options may is two:
- change the relay
- use a transistor to provide the current from the power supply and not from the nodemcu pin