kakopappa/sinric #232: [WSc] Webservice disconnected from sinric.com!

CptZeroG, 2019-03-07:
Hi Y'all,
about some hours I get this error on all my devices:
[WSc] Webservice disconnected from sinric.com!
the WiFi works fine and the connection to other service also works.

It's a  sinric server problem?

kakopappa, 2019-03-07:
I am not sure whether this is problem with the server or Clouldflair or your ISP. Websocket connections go through Clouldflair firewall. Clouldflair seems to reset the ws connections time to time. This is a limition in the free tier. If I change the endpoint iot.sinric.com to skip Clouldflair firewall then the server ip opens up for hackers and DDOS (which we had problems in the past)

check whether it auto reconnect when it disconnects. I think library already handles this.  

Just like taylor swift said in one of her songs, "nothing lasts forever"

Jape1982, 2019-03-10:
Hi,

I have this same issue with Adafruit Huzzah ESP boards. Everything has been working well for a week or so, but yesterday all my devices stopped working. The error that I get is from serial monitor below:

Connecting to Wifi: Koti_XXXX_2GEXT
.....
WiFi connected. IP address: 192.168.XX.XX
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!

Is this a server problem, that is fixed at some time, or what could be the problem ?

Please help, how to fix this ?