kakopappa/sinric #233: ("Turn on unknown device id:");

mascolo74, 2019-03-07:
Sorry for my English,
alexa responds to the commands but on the serial at each command appears the writing:
Serial.print ("Turn on unknown device id:");
  Serial.println (deviceId);
that is, it does not recognize me as a switch and therefore it gives me an unknown switch

mascolo74, 2019-03-07:
void turnOn(String deviceId) {
 if (deviceId == "5cxxxxxxxxxxxxxxxa") // Device ID of device 1
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
  digitalWrite(device1,HIGH);
  delay(500);
  digitalWrite(device1,LOW);
 } 
 else if (deviceId == "5cxxxxxxxxb") // Device ID of device 2
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device2,HIGH);
 delay(500);
 digitalWrite(device2,LOW);
 }
 else if (deviceId == "5cxxxxc") // Device ID of device 3
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device3,HIGH);
 delay(500);
 digitalWrite(device3,LOW);
 } 
 else if (deviceId == "5cxxxxd") // Device ID of device 4
 { 
 Serial.print("Turn on device id: ");
 Serial.println(deviceId);
 digitalWrite(device4,HIGH);
 delay(500);
 digitalWrite(device4,LOW);
 }
 
 else 
 {
 Serial.print("Turn on unknown device id: ");
 Serial.println(deviceId); 
 }

kakopappa, 2019-03-07:
looks like your device id is missing in the if statement. 

check the 
if (deviceId == "5cxxxxd") <---- This should be same as your device id in the dashboard

mascolo74, 2019-03-07:
I rewrote the code 4 times but keep doing the same thing .......... I have 4 pins and the four codes are correct

kakopappa, 2019-03-08:
If you are seeing "Turn on unknown device id: " in the serial terminal, it's likely your device id not in the if else statement. If you can post your code here without the API key i can take a look

mascolo74, 2019-03-08:
I did a complete reset of alexa by removing all the devices, turned off and on alexa and now it works thanks a lot

mascolo74, 2019-03-08:
now I try to make it work with buttons

closed 2019-03-08 by mascolo74