kakopappa/sinric #234: Google Home action submission status

kakopappa, 2019-03-08:
Google Home Action

1. Project submission - Done
2. Policy review Initiated - Done
3. Submission of Compatible Device Form - Done
4. Policy review passed and email confirmation is sent - Waiting
5. Completion of Intake Form (to send devices for testing) - Pending
6. Devices sent for testing and certification (instructions in Policy Email) - Pending
7. Devices pass testing and your project launches! - Pending

@al8943 @servermcp @qwandor @Deimos1994

netamego, 2019-04-24:
Hi,
Any updates about this?. Seems that Amazon have better scheme with their skills.Thanks a lot for your great great work!!!

qwandor, 2019-05-25:
Hi, any update on this?

kakopappa, 2019-05-25:
Yes, Got rejected. 

This is the response I got

One of the issues with your project is how you are having your users interact with your service. We do not want users to follow a process similar to what you’ve listed on your website.  https://github.com/kakopappa/sinric/wiki/How-to-use-Google-Home. 
 
> From my understanding, it appears you created a tutorial for how to allow users to create their own google action using your fulfillment.  This is different from other certified virtual hubs.  
> 
> Thanks,
> 

Then I asked them will it make a difference If I remove it and applied again. No response

Deimos1994, 2019-05-25:
@kakopappa, you probably should explain to them that it is a temporal workaround.
If Google approves then we no longer need to follow that instruction, right?
I suggest adding a warning to that instruction, something like: _"It is a temporary workaround, for beta testers only. After Google approves us we replace it with a new instruction"_

Deimos1994, 2019-05-25:
I think you need to try again. How much time passed since their last response?

saiedmethal, 2019-05-28:
Any update on this issue still can't get to add devices in google home am ia missing some thing or this is it ?

servermcp, 2019-07-05:
Hola, alguna novedad???

netamego, 2019-07-29:
Hi kakopappa,

I have readed that may be you do another attempt with Sinric Pro version (https://github.com/kakopappa/sinric/issues/299#issuecomment-507930494). Are there any clue about when this Pro version will be available and how much will it cost?. 

Thanks a lot kakopappa for your incredible work!!!

Best regards.

kakopappa, 2019-07-30:
@netamego Thank you. 

sorry about the GH. They rejected the action even tough I said it's for testing only. They stopped respoding to my emails after that and rejected the test request. Instead of worring about it, I thought will do a fresh start with pro version

I have been working on the Pro version day and night for some time now. I can't wait for you guys to see it. specially doorbell support :) I will post a video on facebook. Anyways, we are doing the final touches for the website, C++ SDK, Python SDK and Android app now. Amazon Alexa skill is 100% completed now and Google Home action is about 60%. I am planing to launch with Alexa support first and then send two devices to Google for approval. Hopefully by end of August website/alexa/ app  will go live and end of Sep/Oct will have the GH action. 

Pricing haven't decided yet. Thinking of two version free/ paid. Ideas are welcome.