kakopappa/sinric #235: Contact and Motion Sensor

Aggebitter, 2019-03-10:
Any solution on it's way for this ?

https://developer.amazon.com/blogs/alexa/post/bda9d70c-2f0d-454d-9939-2eb82868cf35/announcing-contact-and-motion-sensor-apis-and-integration-into-alexa-routines-adding-new-home-automation-features-for-customers-and-new-opportunities-for-smart-home-developers

// David

kakopappa, 2019-03-10:
Hi, David, 

Contact and Motion Sensor API will be added soon. Thanks

Aggebitter, 2019-03-11:
Can't wait untill I detect my phones MAC-adress and Alexa will respond

gsidirop, 2019-07-28:
Is the motion sensor added to sinric?

kakopappa, 2019-07-28:
@gsidirop I have been testing it for upcoming pro-version. pretty sweet

[image]
[image]

gsidirop, 2019-07-28:
Neat!!! We will be waiting for it. Will be able to trigger an Alexa Routine? I am asking for example because the sinric button SW cannot be used to trugger an Alexa Routine.

maherkheder3, 2019-09-11:
Hi @kakopappa, thank you, it is really very nice.
which time we can use this feature ?