kakopappa/sinric #237: Server does not update device when invoked from API

mitko9000, 2019-03-15:
Hello when using this endpoint https://sinric.com/api/v1/updateDeviceSettings to update the deviceState from ON to OFF the state change is not send to the device, why its so this way ?

kakopappa, 2019-03-15:
What’s the response you got?  likely a parameter issue.

App is also using the same endpoint btw

mitko9000, 2019-03-15:
Well I connected a esp8266 with a sinric and I made a app that communicate with the sinric api and I have a toggle button when I press it it switch from on and off with the above url but the device itself never receive a event that its switched on and that should not be the case because at sincric the status is switched on

kakopappa, 2019-03-15:
Okey now I understand what you are trying to do.

API only update the status on the server database so when Alexa requests it
the server can send the correct status

I can update the api to forward to command to ESP module in one of the
iterations to come.

I will update this issue again when it’s done

mitko9000, 2019-03-15:
The sinric android application does it, what request does it execute ?

kakopappa, 2019-03-20:
sorry about the late response;

here is the code from the android app 
https://github.com/kakopappa/sinric/issues/241

I am not sure why you are developing your own system. Anyway. I am working on a  pro version which supports controlling everything via api and lot more feature. Perhaps you can build a better system when it's ready