kakopappa/sinric #240: Curl command was not working

big-data-elg, 2019-03-20:
When i sent command through curl and API, reply was like below.

{"statusCode":400,"error":"Bad Request","message":"Invalid multipart payload format"}

I use below with my email id and password. 
curl --request POST  --url https://sinric.com/api/v1/login  --header 'cache-control: no-cache'  --header 'content-type: multipart/form-data; boundary=----WebKitFormBoundary7MA4YWxkTrZu0gW'  --form email=<email>  --form password=<password>

kakopappa, 2019-03-20:
I just checked. I can login using Postman without an issue. This is the sample code generated from Postman

curl -X POST \
  https://sinric.com/api/v1/login \
  -H 'cache-control: no-cache' \
  -H 'content-type: application/x-www-form-urlencoded' \
  -H 'postman-token: 2a72c283-06fc-b5e5-9f5a-40979f1e7690' \
  -d 'email=aruna.tennakoon%40gmail.com&password=password'

kakopappa, 2019-03-20:
[image]

kakopappa, 2019-03-20:
[image]

big-data-elg, 2019-03-20:
Thank you for your help.
i got succeed with below at node-red.  :)

msg.headers = {'content-type':'application/x-www-form-urlencoded'};
msg.payload={ email: "xxxxxxx@xxxxxxx", password: "xxxxxxxxxx" };
return msg;

BR,

closed 2019-03-20 by kakopappa