kakopappa/sinric #241: API was not working

big-data-elg, 2019-03-20:
I sent a command through API. and the sinric app changed the status. 
but device was not working.
however, I sent a command through AI speaker. device was working good. 

when I sent command like below. I received "success: true". but device was not working.
msg.headers = {'content-type':'application/x-www-form-urlencoded'};
msg.payload={ action:"setPowerState",value: "ON", deviceId: "xxxxxxxxxxxxxxxxxx", apikey:"xxxxxxxxxxxxxxxxxxxxxx" };
return msg;

kakopappa, 2019-03-20:
Here is the code from android app (written in dart) to turn on/off

`	String powerState =  'ON'; //  'OFF' 'ON';

    Map<String, dynamic> jsonMap = {
      'deviceId': device.id,
      'action' : 'setPowerState',
      'value' : powerState,
      'targetDeviceCmd': 'setPowerState',
      'targetDeviceValue': powerState,
      'commandFormat' : 'alexa' // or 'googlehome'
    };
    
    Map<String, String> headersMap = {
      "Authorization": "Bearer " + authToken
    };
	
    var deviceActionUrl =  "https://sinric.com/api/v2/device/action";
    var data = await NetworkUtil.internal().post(deviceActionUrl, body: jsonMap, headers: headersMap);
      
    if(data["success"]) {
        return true;
    } else {
      return false;
    }
`

big-data-elg, 2019-03-20:
Aha. there was another API. I'll try with it. and Let you know. 
Thank you for your reply. 👍

big-data-elg, 2019-03-21:
My device good works with API "https://sinric.com/api/v2/device/action".
I am using Node-red for handling APIs.

closed 2019-03-21 by big-data-elg

kakopappa, 2019-03-21:
I have seen another Node-red here
https://github.com/mayankraichura/node-red-contrib-sinric-ws

perhaps you can open source some of the work you did ?