kakopappa/sinric #243: increase/decrease volume on TV

Mahmoud-AlQarout, 2019-03-27:
How can I increase/decrease volume using IR transmitter?
Because  "void adjustVolume" give me only one action !

kakopappa, 2019-03-28:
you have to keep the current volume in the memory and adjust it when you receive the command. 

new = current + adjustVolume;

Mahmoud-AlQarout, 2019-03-28:
Thanks for reply 

But I want to send ir code using
Irsend.sendNEC
That will increase the volume
And other one to decrease volume

Because I can't embed together, it will send one of them.

Thanks in advance !

Mahmoud-AlQarout, 2019-03-28:
I solved it .

I used -10 as a value and it's work

            `else if(action == "AdjustVolume") { 
            
            const char* kcvolume = json ["value"]["volume"];
            int volume = atoi(kcvolume);  
            adjustVolume(volume); 

            if (volume == -10){
            irsend.sendNEC(0x718EE21D);
            delay(500);
            irsend.sendNEC(0x718EE21D);
             }
            else {
            irsend.sendNEC(0x718EE01F);
            delay(500);
            irsend.sendNEC(0x718EE01F);
            }
          }`

closed 2019-03-28 by Mahmoud-AlQarout