kakopappa/sinric #245: I have a problem ESP8266WiFi.h: No such file or directory

canuri, 2019-03-30:
Hi,
i search my problem in issues and i found this #60  i tried esp8266wifi.h to wifi.h name change and then i encounter same problem > sinric_deneme1_multi:7:23: error: WiFiMulti.h: No such file or directory

Thats my error code under here
`Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Uno"

sinric_deneme1_multi:6:25: error: ESP8266WiFi.h: No such file or directory

compilation terminated.

exit status 1
ESP8266WiFi.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

kakopappa, 2019-03-30:
Looks like you have not selected the ESp8266 as the board

canuri, 2019-03-30:
i am using arduino uno r3 clone wirh esp8266

canuri, 2019-03-31:
> Looks like you have not selected the ESp8266 as the board
> […](#)

**I selected bord "generic esp8266 module" and result**
Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
sinric_deneme1_multi:223:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

         DynamicJsonBuffer jsonBuffer;

           ^

sinric_deneme1_multi:304:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   DynamicJsonBuffer jsonBuffer;

           ^

sinric_deneme1_multi:320:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   DynamicJsonBuffer jsonBuffer;

           ^

Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\Can\Documents\Arduino\libraries\ESP8266WiFi
 Not used: C:\Users\Can\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi
Multiple libraries were found for "WebSocketsClient.h"
 Used: C:\Users\Can\Documents\Arduino\libraries\WebSockets_2.1.3_549
 Not used: C:\Users\Can\Documents\Arduino\libraries\arduinoWebSockets-2.1.1
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\Can\Documents\Arduino\libraries\ArduinoJson
 Not used: C:\Users\Can\Documents\Arduino\libraries\Antares_ESP8266_HTTP
exit status 1
DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kakopappa, 2019-03-31:
DynamicJsonBuffer is a class from ArduinoJson 5. Please see
arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson
version 6

You have install version 5 ArduinoJson library to solve this compile error

Check the how to install dependencies in the wiki

closed 2019-03-31 by canuri

canuri, 2019-03-31:
i installed version 5 and you solved my problem.
thank you

savitesh, 2020-03-27:
waste