kakopappa/sinric #246: Relay code automatically on

stickerman28, 2019-03-30:
Hi,
I’m using the code example given for a relay, the relay is switched on using LOW and switched off using HIGH.
I’ve changed these in the turn on and turn off sections of the code.
When connecting the device to power, it automatically switches on until Alexa is told to switch it off. Same happens for a power outage, when the power comes back on the device automatically switches on.

How can I amend the code so that the pin is automatically HIGH when the esp8266 boots up in order to keep the device (a lamp) off until Alexa is told to switch it on?

Thanks

kakopappa, 2019-03-31:
You have two solutions here.

1. (ideal) A relay can be  Normally Open/ Normally Closed (NO/NC) by default. Looks like you have wired it wrong. http://3xi5uirb.dietistvanderschaaf.nl/no_nc_relay_diagram.php

2. Set the relay state to turn off in setup()

stickerman28, 2019-03-31:
Thanks.
I am using this relay 
https://m.banggood.com/2-Way-Relay-Module-With-Optocoupler-Protection-p-972428.html?currency=USD&createTmp=1&aff_key=31k3_5rwc1&utm_source=ShareASale&utm_medium=aff&sscid=31k3_5rwc1

temocastro, 2019-05-14:
@stickerman28  could you share your code initialization for the outputs? other we cant help you. The diagram on how you are planning to use the relay will help as well, active high or active low

stickerman28, 2019-05-15:
This solved my issue 
void setup() {
...
pinMode(relayPin1, OUTPUT);
digitalWrite(relayPin1, LOW);
}