kakopappa/sinric #248: All connected devices turn on and off at same time

austingreisman, 2019-03-31:
Hello kakopappa,

I've recently signed up to your service (very easy to by the way) great job on the instructions.
Anytime I go to turn on or off my 3 devices connect (2 lights and 1 fan) they all turn off and on together. It doesn't seem possible to independently control each one. I'm quite sure I've assigned each devices with it's own unique ID since Alexa can discern them when I search for new devices. I'm using separate ESP 8266 01's connected to relays to control them.

Any help would be greatly appreciated.

Thanks,

Austin

kakopappa, 2019-04-02:
Thanks

Check the if  device id comparison logic.  Likely the issue is there or the
 GPIO PIN assignment.

If you still cannot fix it post the code here without your api key

austingreisman, 2019-04-03:
That would make sense... I'll try reflashing the chips and let you know.

Thanks

austingreisman, 2019-04-07:
Hi Kakopappa,

Everything is good now. I didn't realize that when issuing an Alexa command, every separate device is hit with the ID of the device you want. So I just removed my turn off and on else condition for unknown devices and it works now!

Thanks for your help.

Austin

closed 2019-04-08 by kakopappa