kakopappa/sinric #250: help me at code tv

error-dr, 2019-04-02:
i need code to channel up and down tv like that code what should i add to work correctly
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> 
#include <ArduinoJson.h>
// IR part-------------------------------------------------------------
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_LED 4  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(IR_LED);  // Set the GPIO to be used to sending the message.

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "your wifi ssid name" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "your wifi password" // TODO: Change to your Wifi network password

#define API_ENDPOINT "http://sinric.com"
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void TogglePower(String deviceId) {
  if (deviceId == "xxxxxxxxxxxxxxxxxxxxxx") // TODO: Change to your device Id for TV
  {  
     irsend.sendSAMSUNG(0xE0E040BF,32);  // Send a power signal to Samsung TV.
     delay(500);
  } 
  
  if (deviceId == "yyyyyyyyyyyyyyyyyyyyyyy") // TODO: Change to your device Id for Set Top Box
  {  
      
     irsend.sendRC6(0xC0000C,24); // Send a power signal to tatasky.
     delay(500);
  } 
}

void ToggleMute(String deviceId) {
  if (deviceId == "xxxxxxxxxxxxxxxxxxxxxxxxxx") // TODO: Change to your device Id for TV
  {  
     irsend.sendSAMSUNG(0xE0E0F00F,32);  // Send a power signal to Samsung TV.
     delay(500);
  } 
  
  if (deviceId == "yyyyyyyyyyyyyyyyyyyyyyyyy") // TODO: Change to your device Id for Set Top Box
  {  
      
     irsend.sendRC6(0xC0000D,24);  
     delay(500);
  }  

  }

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
           
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { 
            String value = json ["value"];
            if(value == "ON" || value == "OFF" ) {
                TogglePower(deviceId);
                }
        }

        else if(action == "ChangeChannel") {
          String ChannelName=json ["value"]["channelMetadata"]["name"];
          String ChannelNumber=json ["value"]["channel"]["number"];
          

          if(ChannelName=="national geographic"){
            irsend.sendRC6(0xC00007,24);  // Send IR code for remote button 7
           delay(500);
           irsend.sendRC6(0xC00000,24); // Send IR code for remote button 0
           delay(500);
           irsend.sendRC6(0xC00008,24); //Send IR code for remote button 8
           delay(500);
            Serial.println("[WSc] channel: " + ChannelName);
           }
          if(ChannelName=="star movies"){ 
           irsend.sendRC6(0xC00003,24);  // Send IR code for remote button 3
           delay(500);
           irsend.sendRC6(0xC00005,24); // Send IR code for remote button 5
           delay(500);
           irsend.sendRC6(0xC00003,24); // Send IR code for remote button 3
           delay(500);
            Serial.println("[WSc] channel: " + ChannelName);
          }
         }

       
        else if (action == "SetMute") {
          
         bool MuteAction=json ["value"]["mute"];
          if(MuteAction==true || MuteAction==false){ 
             ToggleMute(deviceId);
           }
          
        }

        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
        
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
    irsend.begin();

 WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

kakopappa, 2019-04-02:
I can't remember whether I have added alexa-channelcontroller for tv. What happens when you ask Alexa to chennel up?

https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html

error-dr, 2019-04-02:
i dont add channel up but i add the IR REMOTE code to channel up with voice command alexa change channel to national geographic on tv  and i need to add voice command alexa channel up to my code how do that?

kakopappa, 2019-04-02:
When you ask Alexa to channel up, what happens? Do you see something in
Serial terminal or Alexa says not supported?

error-dr, 2019-04-02:
[WSc] get text: {"deviceId":"5ca0eb9679bd4f2dccdcf0c8","action":"SkipChannels","value":{"channelCount":1}}

kakopappa, 2019-04-02:
There you have it. Now you have to read the channelCount and send up or
down IR code

error-dr, 2019-04-02:
plz can help me to write that code  because i cant

error-dr, 2019-04-02:
can you answer me should i write 
 void channelCount(String deviceId) {
  if (deviceId == "") // TODO: Change to your device Id for TV
  {  
    irsend.sendNEC(0X80BF7B84, 32);  // Send a power signal to Samsung TV.
     delay(500);
  } 
  
  if (deviceId == "yyyyyyyyyyyyyyyyyyyyyyyyy") // TODO: Change to your device Id for Set Top Box
  {  
      
     irsend.sendRC6(0xC0000D,24);  
     delay(500);
  }  

  }

kakopappa, 2019-04-02:
try something like this

`else if(action == "SkipChannels") {
        String  channelCount=json ["value"]["channelCount"];
}`

error-dr, 2019-04-04:
thx it work with me
else if(action == "SkipChannels")
  { String channelCount=json ["value"]["channelCount"]; 
 if(channelCount=="1"){
            irsend.sendNEC(0X80BF7B84, 32);  // Send IR code for remote button 7
           delay(500); }
           if(channelCount=="-1"){
            irsend.sendNEC(0X80BF6B94, 32);  // Send IR code for remote button 7
           delay(500); }
  }
 but some time its conflict when i say channel up on tv the volume up instead channel how can i fix that

error-dr, 2019-04-04:
thx man i solve it by saying next channel and pervious channel instead channel up or down