kakopappa/sinric #252: Devices got disconnected over night

marcosmoura91, 2019-04-07:
Hello, yesterday I connected a nodemcu to the sinric servers and it was working. I also connected a esp01s and they were both working flawlessly.

After I waked up this morning I found out that all my voice commands return an OK but none of them actually changes something. 

I'm using Alexa, and for my devices to work again I had to power them off and on. 

Note:My nodemcu is also a server and the web page was working. Just the connection to sinric was not.

Is there an workaround? I see that the example has a heartbeat function. Can you give me some details here? Should there be a function to try to login after it checks it's not connected anymore?

kakopappa, 2019-04-07:
sometimes connection to the server resets due to errors / your network provider resetting ip / cloudflair firewall. These issues will be fixed in Sinric 2.0. 

For now you have to handle the connection resets. normally websocket library does this and you do not require any special coding. make sure to send the heartbeat and your ESP does not go to sleep

michaelfisc, 2019-04-30:
> sometimes connection to the server resets due to errors / your network provider resetting ip / cloudflair firewall. These issues will be fixed in Sinric 2.0.
> 
> For now you have to handle the connection resets. normally websocket library does this and you do not need any special coding. make sure to send the heartbeat and your ESP does not go to sleep

so hyped for sinric 2.0