kakopappa/sinric #254: thermostat

avelas1, 2019-04-09:
First Of all thank you for helping and working for your contributions. I downloaded and installed the code thermostat_example_with_dht_11. Ino. I'Ve had problems with the operation of the code. I've installed It on an R1 WEMOS D1 board. It Doesn't Show me the temperature of the sensor. I Have changed the code to use in a NodeMCU with a BME280 and the same. 

You Have A reference to the Variables and procedures or failing code description

Thank you

kakopappa, 2019-04-10:
thanks. 

It showes the tempreature in Alexa app only (not anywhere else).  

make sure you are sending the correct device id here.
https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example_with_dht_11.ino#L187

are you sending minus (-) tempreature values ?

anyways, here what it looks like
https://www.facebook.com/sinrichome/photos/a.353020545282550/353020335282571/?type=3&theater

Sinric 2.0, will have a better API / tempreature graph and an App.. It will be released in couple of weeks from now

avelas1, 2019-04-10:
Thank you very much for your help and your clarifications!

acvelasco2@gmail.com

Antes de imprimir este mensaje, asegúrate de que es necesario. Proteger el
medio ambiente está también en tu mano.

The-Great-One, 2019-06-11:
setSetTemperatureSettingOnServer("<device id>", temperature, "FAHRENHEIT", temperaturefh, humidity); 
In this line do is replace <device id> with the device id of my thermostat as shown on sinric.com? And is there any place else what changes do i have to make to trigger a relay based on set temperature in the alexa app

kakopappa, 2019-06-12:
> setSetTemperatureSettingOnServer("", temperature, "FAHRENHEIT",
> In this line do is replace with the device id of my thermostat as shown on

This updates the temperature values the server report to Alexa when
requested

Turing on a relay when certain temperature has met must be done in ESP.

Rules and actions will be part of upcoming pro version

closed 2019-06-12 by avelas1