kakopappa/sinric #255: this is not an issue but a question?

pico563, 2019-04-11:
Hello Kakopappa
I am asking if it is possible to add Sinric to Tasmota FW as many of us are using Tasmota now.
the limit with  Tasmota is that it can control only a single device using belkin wemo. 

[https://user-images.githubusercontent.com/5904370/53055478-24ea5d80-34a8-11e9-8f5e-17f4ef4c4fd3.png](url)

thanks

pico563, 2019-04-12:
I know it can be done but don't know how to begin.

kakopappa, 2019-04-12:
I am not familar with Tasmota FW. Never used it.. I dont think it can be that hard. Just need to copy the same code and add websocket library.

pico563, 2019-04-12:
thanks kakopappa for your response.
I like Tasmota because it is made by experts and you can have solution for any project in your mind.
I hope a day you can take a closed look to it.
thanks.

closed 2019-04-12 by pico563