kakopappa/sinric #256: Alexa cannot find device

viperchannel, 2019-04-13:
Hi! I uploaded switch_example sketch to a Node MCU and it recieves commads via the sinric web interface. 

`WiFi connected. IP address: 192.168.178.56
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] get text: {"action":"test","sender":"web"}
[WSc] received test command from sinric.com

I also enabled the sinric skill in my amazon account, linked it using my siniric email and password (recieved sucess message), but I cannot find my device either via the amazon web interface, or the "find devices" voicecommand on my echo (1st gen) or the alexa phone app. 

Any ideas what I could do? Thanks!

kakopappa, 2019-04-13:
That’s strange. Normally it works if you use the same account when linking
the Alexa skill.

Delete all the devices.
Create a simple switch
Unlinking and link again
Discover

viperchannel, 2019-04-13:
Hi, I already created a new sinric account with just one device on it. -a switch- and reflashed the code with the new API and device code. 

Unlinked and linked the skill several times. (with the new account info)

Same as before, on the sinric website, I see my connected IOT device and in the serial monitor, I see the commands coming from the website. 

But finding it via alexa- 
No luck.

Akaya16, 2019-04-13:
Same exact problem here. I can turn the Relay on and off via the sinric app but my Echo can't find any device

kakopappa, 2019-04-13:
OK - Give me an hour or two. I will login to server and check what's going
on

yamen01, 2019-04-13:
same for me ! thanks for take care of it !

viperchannel, 2019-04-13:
Great. Let me know if you need any additional information. Both my accounts have viperchannel @ in the emailadress, in case you want to check them out

kakopappa, 2019-04-13:
some dude in Sweden flooding the server
with "action.devices.commands.OnOff" commands

malhotrasujeet, 2019-04-13:
Sinric Handshake fail.... unable to link

kakopappa, 2019-04-13:
yes. i am doing some testing on the server. check again in few mins

malhotrasujeet, 2019-04-13:
Okay Thanks.

viperchannel, 2019-04-13:
Please let me know when I can try connecting it and thanks alot for your work!

viperchannel, 2019-04-15:
Now my Alexa app found the device. - But the device itself keeps printing
 [WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
[WSc] Webservice disconnected from sinric.com!
in the serial monitor

kakopappa, 2019-04-15:
Make sure your API key is correct

And What’s your ip?

viperchannel, 2019-04-15:
I just ran into a similar problem with another wifi device in my network. - So this connection issue may not be caused by sinric. - I will first check this out and report back.

viperchannel, 2019-04-15:
Hey, it seems to be working now. It was some problem with my WiFi which caused that my NodeMCU as well as my other IOT NodeMCU and my Amazon and TPLink smarthome devices were unable to connect to the internet. I have no idea what that was, I had to completely reset my router and now it is all working great again

yamen01, 2019-04-15:
sorry but for me i still have some trouble last week i was able to add a device but since alexa do not find any new device ...

yamen01, 2019-04-15:
i have a question ( alexa looking for the chips or just for sinric for add a switch ?) ( am a noob )

yamen01, 2019-04-17:
work so well now ! am happy as fuck !

kakopappa, 2019-04-18:
There was an issue here as well. If you don’t put a description for the device when you create one, Alexa stops showing it in the app. I fixed it last night

viperchannel, 2019-04-18:
Cool, thank you! I also didn't put in a description but on my last device I did it for some reason. This is an awesome project! And what I also really like about it, is that the device listens to the Server and the commands also get send to the server. So it is basically cloud based. This is great for users that only get a IPv6 adress from the provider and share an IPv4. This is the case in some countries and I really had trouble reaching other devices in my homenetwork from outside, since my phone only gets a IPv4. That needs tunneling and other stuff to get it working.

closed 2019-04-18 by viperchannel