kakopappa/sinric #257: SINRIC performance problem,

ecuille, 2019-04-14:
I have a performance problem, often SINRIC takes a while to respond , 
I have to try 2 or 3 times to get an answer from SINRIC

MickTheGinge, 2019-04-14:
Issue seemed to start today April 14th. Alexa interface is reporting "Device is unresponsive" Sinric website interface laggy too.

kakopappa, 2019-04-15:
today I got the "Device is unresponsive" as well. The reason is some people
are flooding the server with on/off and set temperature requests. I will
implement a rate-limit for the API and websocket server

MickTheGinge, 2019-04-15:
Hi kakopappa
What is the recommended minimum interval? I update temperature values once every 30 seconds, x4 sensors, and a couple of switches status every 2 seconds. If I wait longer for the switches the Alexa app interface seems too laggy. I look forward to your recommendation. 

You have created an amazing interface, please keep up the great work.

kakopappa, 2019-04-15:
Thanks.

Temperature - every minute
Switches  - when the status changes.  send it once.

Sending status changes to server every 2 seconds is too short. it just
keeps updating the server database with the same status over and over again

MickTheGinge, 2019-04-15:
Perfect. After I wrote the note above I suddently asked myself why I was updating the switch like that, state change is far more efficient. I was being lazy, sorry

ecuille, 2019-04-15:
thanks, it looks more stable

MickTheGinge, 2019-04-16:
Issue seems back today. April 16th 13:00 PST 2019

ecuille, 2019-04-18:
Hi Kakopappa

The Issue persist, I have ti try 2 o 3 times to get answer from SINRIC

What Can I do?

kakopappa, 2019-04-18:
Normally this happens when the server doesn’t respond back to Alexa on
time.

someone must be flooding the server again with on/off requests. (Keeps the
server busy) I will take a look and block the ips and ask them to fix the
code

Long term solution to this problem is

1. Upload a code example for push button.

2. Update the server logic with request rate limit and update the server
code to run in a cluster.

I will let you know when this is done.

kakopappa, 2019-04-18:
@ecuille are you still having issues?

1. - done

ecuille, 2019-04-20:
Hi

Yes. 21:46 mexico time, the SINRIC no response  at first Call.
I have to try 2 or 3 times

kakopappa, 2019-04-20:
load on the server seems to be normal. .. 

are you still around?

ecuille, 2019-04-20:
[image1.png]
Hi

I try 10 times only in one returned  error .

Regards

kakopappa, 2019-04-20:
Yes. Seems to happen often now.

MickTheGinge, 2019-04-26:
Hi
   I’d like more information on Sinric PRO please. Features, pricing etc. 

So far I’m only using the Sinric connection for my personal devices but I might be expanding in low volumes enabling others in the future. 

The Alexa interface is a nice to have, my main capabilities are built around IFTTT so I’ve not yet decided if I market the Alexa control. In addition to IFTTT My device has a full web interface but the Alexa GUI is so much cleaner and of course makes the device feel more valuable. 

Either way I owe you a donation for your efforts. 

Best regards 

Mick Posner