kakopappa/sinric #258: sinric setdevice api bad gateway

ImJustChew, 2019-04-16:
<span class="cf-error-type">Error</span>
                                    <span class="cf-error-code">502</span>
                                    <small class="heading-ray-id">Ray ID: 4c846192d8aba227 &bull; 2019-04-16 07:17:44 UTC</small>
                                </h1>
                                <h2 class="cf-subheadline">Bad gateway</h2>
                            </div>
                            
                            <div class="cf-section cf-highlight cf-status-display">
                                <div class="cf-wrapper">
                                    <div class="cf-columns cols-3">
                                        <div id="cf-browser-status" class="cf-column cf-status-item cf-browser-status ">
                                            <div class="cf-icon-error-container">
                                                <i class="cf-icon cf-icon-browser"></i>
                                                <i class="cf-icon-status cf-icon-ok"></i>
                                            </div>
                                            <span class="cf-status-desc">You</span>
                                            <h3 class="cf-status-name">Browser</h3>
                                            <span class="cf-status-label">Working</span>
                                        </div>
                                        <div id="cf-cloudflare-status" class="cf-column cf-status-item cf-cloudflare-status ">
                                            <div class="cf-icon-error-container">
                                                <i class="cf-icon cf-icon-cloud"></i>
                                                <i class="cf-icon-status cf-icon-ok"></i>
                                            </div>
                                            <span class="cf-status-desc">Seoul</span>
                                            <h3 class="cf-status-name">Cloudflare</h3>
                                            <span class="cf-status-label">Working</span>
                                        </div>
                                        <div id="cf-host-status" class="cf-column cf-status-item cf-host-status cf-error-source">
                                            <div class="cf-icon-error-container">
                                                <i class="cf-icon cf-icon-server"></i>
                                                <i class="cf-icon-status cf-icon-error"></i>
                                            </div>
                                            <span class="cf-status-desc">sinric.com</span>
                                            <h3 class="cf-status-name">Host</h3>
                                            <span class="cf-status-label">Error</span>
                                        </div>
                                    </div>
                                </div>
                            </div>
                            
                            <div class="cf-section cf-wrapper">
                                <div class="cf-columns two">
                                    <div class="cf-column">
                                        <h2>What happened?</h2>
                                        <p>The web server reported a bad gateway error.</p>
                                    </div>
                                    <div class="cf-column">
                                        <h2>What can I do?</h2>
                                        <p>Please try again in a few minutes.</p>
                                    </div>
                                </div>
                            </div>
                            
                            <div class="cf-error-footer cf-wrapper">
                                <p>
                                    <span class="cf-footer-item">Cloudflare Ray ID: 
                                        <strong>4c846192d8aba227</strong>
                                    </span>
                                    <span class="cf-footer-separator">&bull;</span>

kakopappa, 2019-04-16:
can you add more details/steps ?

ImJustChew, 2019-04-16:
this is the code I use to trigger the event:
`const formData = {
    "action": 'setPowerState',
    "value": state,
    "deviceId":deviceIds[devID],
    "apikey":token
  };
  request.post({url:'https://sinric.com/api/v1/updateDeviceSettings', formData: formData}, function optionalCallback(err, httpResponse, body) {
    try{
      body = JSON.parse(body);
      if (err) {
        return console.error('upload failed:', err);
      }
      if(body.success){
        console.log('Setting Device ' + devID +' Successful ' + moment().format('MMMM Do YYYY, h:mm:ss a'));
      }
    } catch(e){
      if (e instanceof SyntaxError) {
        console.log(e);
    } else {
        console.log(e);
    }
    }
  });`

and it returns 

Error FoundError: Unexpected server response: 500
Disconnected from IoT System for 1006 because

kakopappa, 2019-04-16:
have you tried using post man?

kakopappa, 2019-04-16:
```
var request = require("request");

var options = { method: 'POST',
  url: 'https://sinric.com/api/v1/updateDeviceSettings',
  headers: 
   { 'Postman-Token': 'a0fe8ac2-6c28-42ab-97d6-8e296d12f764',
     'cache-control': 'no-cache',
     'content-type': 'multipart/form-data; boundary=----WebKitFormBoundary7MA4YWxkTrZu0gW' },
  formData: 
   { action: 'setPowerState',
     value: 'OFF',
     deviceId: 'xxxxx',
     apikey: 'xxxxxx' } };

request(options, function (error, response, body) {
  if (error) throw new Error(error);

  console.log(body);
});

```

kakopappa, 2019-04-16:
just tested this and it works

ImJustChew, 2019-04-16:
Yes, I used postman and I got the bad gateway return on both.

kakopappa, 2019-04-16:
what's your ip ?

kakopappa, 2019-04-16:
[image]

ImJustChew, 2019-04-16:
It's working again, it works but i just encounter problems a lot more often. I'll send you the ip of this happens again.