kakopappa/sinric #263: sinric experimental o real

Eugeniobcn, 2019-04-24:
hola de nuevo, estoy muy  interesado en sinric, quiero usarlo para proyectos grandes, pero me surge una duda, 
sinric, es una aplicación experimental, para que juguemos con nuestros dispositivos de fabricación casera? o por el contrario,
sinric es algo real que perdurara en el tiempo? 
no me importa pagar alguna cuota si con ello puedo ofrecer un producto de calidad y estable, gracias

kakopappa, 2019-04-25:
hi, thanks. 

Sinric has been around over an year now. I will try to keep this project live as much as i can. When i started this project, i never thought it will have 900-1000 devices connected. I just wanted something for self and then i decided to open it forevery one so we all clould benefit. Right now Sinric is running in a small VPS sevrer and payments for the VPS and Amazon lambada expression are made from my pocket and donations. This is not viable with growing no of users and devices everyday. So I have started working on a Sinric pro version. With the pro version you can sign up as a device manufacture or a normal user (like now). It has a better API, an app and high-availability (nodejs cluster). If you guys want to give it a try let me know. 

Hola Gracias.

Sinric ha estado alrededor de más de un año. Intentaré mantener este proyecto en vivo tanto como pueda. Cuando comencé este proyecto, nunca pensé que tendría 900-1000 dispositivos conectados. Solo quería algo para mí mismo y luego decidí abrirlo para que todos pudiéramos beneficiarnos. En este momento, Sinric se está ejecutando en un pequeño servidor de VPS y los pagos por el VPS y la expresión lambada de Amazon se realizan desde mi bolsillo y donaciones. Esto no es viable con un número creciente de usuarios y dispositivos todos los días. Así que he empezado a trabajar en una versión profesional de Sinric. Con la versión Pro puedes registrarte como fabricante de un dispositivo o usuario normal (como ahora). Tiene una mejor API, una aplicación y alta disponibilidad (cluster de nodejs). Si quieren probarlo, avísenme.

Eugeniobcn, 2019-04-25:
por supuesto que quiero probarlo, tengo grandes proyectos en mente, no quiero  controlar  simplemente la persiana de mi casa o el riego del jardín.
Estoy diseñando electrónica para controlar la casa entera, hoteles restaurantes, mi electrónica sera modular similar a domotica SIEMENS, HAGER,, pero utilizando mecanismos analógicos, no digitales, abaratando el coste de la instalación un 80% del precio habitual, haciendo posible que cualquiera pueda tener domotica en casa, 
me gustaría ver la nueva versión y ver si se ajusta   a lo que estoy buscando.
Gracias

kakopappa, 2019-04-25:
Ok. drop me your email address here. I will send you more details via email later today. 

De acuerdo. Déjame tu dirección de correo electrónico aquí. Te enviaré más detalles por correo electrónico más tarde hoy.

Eugeniobcn, 2019-04-25:
ok, eugeniobcn@hotmal.com
Gracias

MickTheGinge, 2019-04-26:
Drop me a note too please? mick at mickandsarah dot com

kakopappa, 2019-04-26:
Sure. I will send you an email in few minutes

qwandor, 2019-05-25:
Can you tell me more about the pro version too please? I'm qwandor@gmail.com.

jpnos26, 2019-06-19:
Hi send me information for pro at athos at dhtlog dot it

israeljaimeshernandez, 2019-08-21:
Hola,

Estoy interesado en la versión profesional si me puedes mandar información estoy iniciando proyecto de domotica  en México. israeljaimeshernandez@gmail.com

Saludos