kakopappa/sinric #264: add a physical button to the sketch

pietrozanc, 2019-04-26:
hi, first of all, I wanted to thank you for the fantastic work you have done very well. now I wanted to create a sketch that handles 4 relays controlled by alexa and by 4 physical buttons, the 4 relays with alexa work really well, but I don't know how to add the physical button. I state that is a little time that I use arduino, I tried with the sketch switch_with_push_button.ino but I didn't succeed,someone could help me to understand what I have to insert in the sketch?

kakopappa, 2019-04-26:
Create a new sketch with 4 push buttons using the AceButton library without
Sinric

https://github.com/bxparks/AceButton

After that it’s easy to integrate

pietrozanc, 2019-04-26:
thanks I managed to do something now i can turn on the relay whit the button but now the problem how to turn off the relay with the same button (I need to understand how to use the button as a switch, once to turn on the other to turn off)

kakopappa, 2019-04-27:
try this

`

#include <AceButton.h>

using namespace ace_button;

const int RELAY_PIN = 2;

const int BUTTON_PIN = 2;
const int RELAY_ON = HIGH;
const int RELAY_OFF = LOW;

AceButton button(BUTTON_PIN);

void handleEvent(AceButton*, uint8_t, uint8_t);

void setup() {
  delay(2000);

#if defined(ARDUINO_AVR_LEONARDO)
  RXLED0; // LED off
  TXLED0; // LED off
#endif

  pinMode(RELAY_PIN, OUTPUT);
  pinMode(BUTTON_PIN, INPUT_PULLUP); // assuming you don't have a resistor connected
  button.setEventHandler(handleEvent);
}

void loop() {
  button.check();
}

void handleEvent(AceButton* /* button */, uint8_t eventType,
    uint8_t /* buttonState */) {
  switch (eventType) {
    case AceButton::kEventPressed:
      digitalWrite(RELAY_PIN, RELAY_ON);
      break;
    case AceButton::kEventReleased:
      digitalWrite(RELAY_PIN, RELAY_OFF);
      break;
  }
}
`

pietrozanc, 2019-04-27:
I tried it, and tried other similar sketches, but the output goes off every time I leave the button. I need a way that the button, once pressed, turns the output on then remains on and the next time tha i press the button the output goes off..

thanks again pietro

pietrozanc, 2019-04-29:
ci sono riuscito utilizzando questo sketch ed integrandolo poi a sinric, ora però non riesco a farlo con altri 3 pulsanti e le altre 3 rispettive uscite :(

This is the sketch with only one button and it WORK, if someone needed it! :)

```

#include <AceButton.h>

using namespace ace_button;

const int Relay1 = 5;
const int BUTTON_PIN = 14;
bool StRelay1 = false;

AceButton button(BUTTON_PIN);

void handleEvent(AceButton*, uint8_t, uint8_t);

void setup() {
  delay(2000);

  pinMode(Relay1, OUTPUT);
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  button.setEventHandler(handleEvent);

  ButtonConfig* buttonConfig = button.getButtonConfig();
  buttonConfig->setEventHandler(handleEvent);
  buttonConfig->setFeature(ButtonConfig::kFeatureClick);
}

void loop() {
  button.check();
}

void handleEvent(AceButton*, uint8 eventType, uint8) {
  switch (eventType) {
    case AceButton::kEventClicked:
      processClicked();
      break;
  }
}

void processClicked() {
  if (StRelay1) {
    digitalWrite(Relay1, LOW);
    StRelay1 = false;
  } else {
    digitalWrite(Relay1, HIGH);
    StRelay1 = true;
  }
}
```

pietrozanc, 2019-04-29:
I fixed, thanks kakopappa!  if someone needs the code I can post it

closed 2019-04-29 by pietrozanc

satbranigan, 2019-05-03:
Hello, I am interested in this code, could you put the complete code and do you know how to do a pulse of 2 seconds on the led?
thanks

kakopappa, 2019-05-03:
There is a sample code already available in GitHub with push button

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino#L29

Use the delay(2000); to pause for two seconds

temocastro, 2019-05-20:
Alexa cant use this button for trigger routines, need help

philb1192, 2019-08-01:
Hi Pietrozanc, I want to build a similar project as yours, mine will only use one relay controlled with the ESP8266 and an external button to manually turn on and off a lamp and update the status into Sinric. I've tried the ''switch_with_push_button.ino'' sketch but the relay output doesn't change states. Is there something in the sketch I should add beside the ApiKey, SSID, Wifi password and Device ID?  I'm fairly new with the Arduino and I'm trying my best to find the issue.

kakopappa, 2019-08-01:
For some it doesn’t work because they send the device name instead of
device id. I have seen this happening and this used to crash the server

This is one of the reasons why we are working on a SDK for Pro. Sometimes
we make small mistakes and it can be frustrating to diagnose. A SDK could
handle these automatically

philb1192, 2019-08-01:
Okep thank you.
But the thing I don't understand is that the output pin is not defined in the sketch. Which pin will make the action of turning on and off the relay?

kakopappa, 2019-08-02:
@philb1192 Pins are upto you, You can connect your module to an any PIN you like.
 https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino#L29 is using D1. D1 contant is only available in Wemos D1 ESP modules (i think). 

Look at your board PIN layout and connect to a PIN and change it here

philb1192, 2019-08-02:
@kakopappa Okep! But this is the input pin for the push button. I was wondering about the output pin that will changes states when the device is turn on and off on the alexa app.

kakopappa, 2019-08-02:
@philb1192 https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_with_push_button.ino#L169

this line sends the status to the server. When Alexa app asks the sinric server what's the status it will return the status from here.

killerpool09, 2020-06-22:
> hi, first of all, I wanted to thank you for the fantastic work you have done very well. now I wanted to create a sketch that handles 4 relays controlled by alexa and by 4 physical buttons, the 4 [...]

> I fixed, thanks kakopappa! if someone needs the code I can post i

> try this
>
> `
>
> #include <AceButton.h>
>
> using namespace ace_button;
>
> const int RELAY_PIN = 2;
>
> const int BUTTON_PIN = 2;
> const int RELAY_ON = HIGH;
> const int RELAY_OFF = LOW;
>
> AceButton [...]

Can you help me to add 3 more physical buttons with the sketch of switch_with_push_button.ino

/*
 Version 0.1 - April 18 2019
*/ 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

#include <AceButton.h> // https://github.com/bxparks/AceButton
using namespace ace_button;

int TEST1 = 16;
int TEST2 = 5;
int TEST3 = 4;
int TEST4 = 0;

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "d9bcd546-2334-4af3-b2e7-d50317676927" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "pro not noob" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "9879164669" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

// Switch
AceButton button1(14);
AceButton button2(12);
AceButton button3(13);
AceButton button4(15);

void button1Handler(AceButton* button1, uint8_t eventType, uint8_t buttonState) {
  Serial.println("button1");
}

void button2Handler(AceButton* butto2, uint8_t eventType, uint8_t buttonState) {
  Serial.println("button2");  
}

void button3Handler(AceButton* button3, uint8_t eventType, uint8_t buttonState) {
  Serial.println("button3");
[... 227 lines cut ...]
      break;
  }
}
void button3Handler(AceButton* button3, uint8_t eventType, uint8_t buttonState) {
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer("<device id>", "ON");
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      break;
  }
}
void button4Handler(AceButton* button4, uint8_t eventType, uint8_t buttonState) {
  switch (eventType) {
    case AceButton::kEventPressed:
      Serial.println("kEventPressed");
      setPowerStateOnServer("<device id>", "ON");
      break;
    case AceButton::kEventReleased:
      Serial.println("kEventReleased");
      break;
  }
}

tusharsahni22, 2020-06-27:
Killerpool09 hello can you plz help me with code 

I want to make 8 channel relay with touch button as well as voice control commands with alexa plz help me with the code

killerpool09, 2020-06-28:
> Killerpool09 hello can you plz help me with code
> 
> I want to make 8 channel relay with touch button as well as voice control commands with alexa plz help me with the code

Currently, even I'm trying to figure out how to add more switches in the code "switch_with_push_button.ino"..

tusharsahni22, 2020-06-28:
> > Killerpool09 hello can you plz help me with code
> > 
> > I want to make 8 channel relay with touch button as well as voice control commands with alexa plz help me with the code
> 
> Currently, even I'm trying to figure out how to add more switches in the code "switch_with_push_button.ino"..

Brother can you send me code which have  4 or 5 switch with alexa voice commands

killerpool09, 2020-06-28:
> Brother can you send me code which have 4 or 5 switch with alexa voice com

YUP, in an hour

tusharsahni22, 2020-06-28:
> YUP, in an hour

Okay brother i am in very confusing about how the switch and the relay work with alexa voice commands and touch switch plz send me code for highest no. of switch that work together. 
Thanks

killerpool09, 2020-06-28:
> Okay brother i am in very confusing about how the switch and the relay work with alexa voice commands and touch switch plz send me code for highest no. of switch that work together.
> Thanks

https://www.youtube.com/watch?v=WZU0P7Gtd_w CHECK IT OUT, BRO

tusharsahni22, 2020-06-28:
> https://www.youtube.com/watch?v=WZU0P7Gtd_w CHECK IT OUT, BRO

Brother i have already watched this type of video but my problem is that i want to connect physical touch button also that can also work to control relay with the alexa commands

tusharsahni22, 2020-06-28:
> Brother i have already watched this type of video but my problem is that i want to connect physical touch button also that can also work to control relay with the alexa commands

Hey bro did you understand my problem i already did alexa control relay but now i want to install physical button with the same relay with alexa voice control

killerpool09, 2020-07-03:
> Hey bro did you understand my problem i already did alexa control relay but now i want to install physical button with the same relay with alexa voice control

Current, I'm still stuvk in feedback situation, if you want to try something by yourself, here's my code(still not finished). 
[Multiple physical feedback button trial.txt]

https://github.com/kakopappa/sinric/files/4869538/Multiple.physical.feedback.button.trial.txt

tusharsahni22, 2020-07-04:
> Current, I'm still stuvk in feedback situation, if you want to try something by yourself, here's my code(still not finished).
> [Multiple physical feedback button trial.txt] [...]

Can you explain the problem in this code