kakopappa/sinric #267: Help with Sirinc Switch Example

Wanganator414, 2019-05-05:
Hello, I am currently trying to control a servo using NodeMcu (Cp210x model) with Amazon Alexa. I am working with the Arduino switch example in the Sinric master folder. I have followed instructions so far to install the non-beta versions of ArduinoJson but ran into a roadblock with the WebSocketsClient.h library. I downloaded the zip files suggested in the example and got the lib ArduinoWebSockets, which is a different lib than the one in the example (WebSocketsClient.h). I am now unsure how to proceed after this step. Any guidance would be appreciated, please let me know if any additional information is needed to clarify the issue (First time posting an issue).

Thanks.

Dhanush4444, 2019-05-06:
@Wanganator414 Try downloading ArduinoJSON of version 5.12.+ from [here](https://www.arduinolibraries.info/libraries/arduino-json). And also try lowering the version of WebSocketClient. [websocket](https://www.arduinolibraries.info/libraries/web-sockets)

Wanganator414, 2019-05-07:
Hello Dazz, thanks for the info. Lowering WebSocketClient version worked for me.

closed 2019-05-07 by Wanganator414