kakopappa/sinric #277: Google Home + Alexa in same program, same ESP8266

temocastro, 2019-05-13:
Hello,

I have one Alexa-ESP8266 and one GoogleHome-ESP8266, I would like to have both in the board/program, I notice they look basically the same, I just can't make it work,

Any idea/help you can provide? an example?

Thanks

dazzHere, 2019-05-13:
@temocastro You want both features in same board/program?

temocastro, 2019-05-13:
yes, same as any WEMO or TP-LINK device that respond to any alexa or google home, same device, 2 different inputs (alexa or google home)

dazzHere, 2019-05-13:
Unfortunately we cannot do that! Because as it's a websocket sever, you authorize to server using a single API at once. You cannot use 2 API to authorize from single instance.

temocastro, 2019-05-14:
I would guess same API, you will connect to Sinric momentary for change the state, correct?

dazzHere, 2019-05-14:
Mail me your code let me check. dhanushdazz@gmail.com

temocastro, 2019-05-14:
done

temocastro, 2019-05-14:
Success, I was able to control my single ESP8266 with Google and Alexa

Just the payload needs to change, so I modified the Google Home sketch and added the Alexa payload condition for "Action"

Like this

....
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "action.devices.commands.OnOff") { // Switch 
            String value = json ["value"]["on"];
            Serial.println(value); 
            
            if(value == "true") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        **else if (action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }**
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
....

temocastro, 2019-05-14:
I will leave this open few days, in cased you have questions, I would like to upload the example, so everybody can use it in the future

@kakopappa take a look, let me know if this makes sense

kakopappa, 2019-05-14:
yep. looks correct.

closed 2019-05-14 by temocastro