kakopappa/sinric #279: cannot use mute fonction with script

dedelajoie, 2019-05-15:
I created a TV device type through Sinric
I start the script "tv_example.py"
When giving orders to Alexa, it works on "ON" and "OFF" orders but when asking to MUTE the device, an error is raised : 

error from callback <function on_message at 0x7f6e5489b758>: malformed string
  File "/home/stp/.local/lib/python2.7/site-packages/websocket/_app.py", line 345, in _callback
    callback(self, *args)
  File "tv_example.py", line 78, in on_message
    obj = ast.literal_eval(message)
  File "/usr/lib/python2.7/ast.py", line 80, in literal_eval
    return _convert(node_or_string)
  File "/usr/lib/python2.7/ast.py", line 63, in _convert
    in zip(node.keys, node.values))
  File "/usr/lib/python2.7/ast.py", line 62, in <genexpr>
    return dict((_convert(k), _convert(v)) for k, v
  File "/usr/lib/python2.7/ast.py", line 63, in _convert
    in zip(node.keys, node.values))
  File "/usr/lib/python2.7/ast.py", line 62, in <genexpr>
    return dict((_convert(k), _convert(v)) for k, v
  File "/usr/lib/python2.7/ast.py", line 79, in _convert
    raise ValueError('malformed string')

I know almost nothing in Python and if I get there is a problem with the format I am not able to correct it. 

thank you for reading

dazzHere, 2019-05-15:
@dedelajoie I will check it out. Just a minute

dazzHere, 2019-05-15:
This error occurred because `ast.literal_eval()` had problem converting nested `dict string` to `dict`. json library solved this problem. [Check out this updated code](https://github.com/dazzHere/sinric)

dedelajoie, 2019-05-16:
thank you so much