kakopappa/sinric #281: Esp8266 unresponsive

schaffertown, 2019-05-16:
My esp8266s become unresponsive and I have to reset them. I have 2 of them on Alexa
. If I reset one only when they are both working it will work and the other will be first to fail. If I reset both at the same time they usually fail at the same time. It seems to be related to power cycling my modem and router. What do you think is the problem? What affect would changing the heartbeat have? Would a watchdog timer help?
Thank you for a wonderful service you have provided.

dazzHere, 2019-05-17:
@schaffertown Why are you using 2 esp8266?

kakopappa, 2019-05-17:
well, I have few devices at home working all the time so I assume your devices lose connection to the server at one point. 

Easy solution would be to reset the ESPs in certian time intervals. Will that fix your problem?

schaffertown, 2019-05-17:
I am using 2 esp8266 to work 16 relays. Yes I would like to reset the boards at 3am eastern time, that should solve the problem. What should I add to the code? Thank you.

kakopappa, 2019-05-17:
try this library
https://github.com/PaulStoffregen/TimeAlarms